LumaTeam/Luma3DS #590: Crash when installed boot9strap 1.2 and copy luma 8.0 to SD and open new 3ds on V9.1.0-20J

wang80919, 2017-06-11:
**System model:**
new 3ds

**SysNAND version (+emu/redNAND version if applicable):**
9.1.0-20J SysNAND
11.4.0.37J emuNAND

**Entrypoint (How/what you're using to boot Luma3DS):**
Boot9Strap

**Luma3DS version:**
8.0
[e.g. 7.1 stable or if using nightly/hourly specify the commit like this https://github.com/AuroraWright/Luma3DS/commit/9570e6cbeca53128433abbf5e3473cb8a07fe69e]

**Luma3DS configuration/options:**

Default EmuNAND: ( )
1
Screen brightness: ( )
4
Splash: ( )
off
PIN lock: ( )
off
New 3DS CPU: ( )

--
off
Autoboot EmuNAND: ( )
off
Use EmuNAND FIRM if booting with R: ( )
on
Enable loading external FIRMs and modules: ( )

off
Use custom path: ( )
off
Enable game patching: ( )
off
Show NAND or user string in System Settings: ( )
on
Show GBA boot screen in patched AGB_FIRM: ( )
off
Patch ARM9 access: ( )
off
Set developer UNITINFO: ( )
off
--

**Explanation of the issue:**

**Steps to reproduce:**

1.
use Homebrew Launcher ot install boot9starp ok
2.
copy lunm 8.0.7 to sd
3.
open 3ds
4.
sysNAND loading crashed.
enuNAND 11.4.0.37J loaded ok.

**Dump file:**

wang80919, 2017-06-11:
add dujp file

wang80919, 2017-06-11:
问题可能的原因：
我以前是 MenuHax
解决办法：
找到对应版本的 menuhax_manager
copy to sd \3ds\
用 Soundhax 启动 Homebrew Launcher
启动 menuhax_manager
选择 delete
然后 重新 安装 boo9strap

wang80919, 2017-06-11:
我自己关。

closed 2017-06-11 by wang80919

jonirrings, 2017-06-11:
……自己人啊。
帮你翻译一下：
### possible cause:
used to be MenuHax
### solution:
1. find the corresponsive  version of menuhax_manager
2. copy it to SD/TF root `\3ds`
3. boot into HBL with soundhax
4. boot menuhax_manager
5. delete menuhax_manager
6. re-install boot9strap