LumaTeam/Luma3DS #591: Random Red Screen of Death when exiting Friend List

SvenDaHacker64, 2017-06-11:
**System model:**

New Nintendo 3DS XL

**SysNAND version (+emu/redNAND version if applicable):**

11.4.0-37E, sysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap v1.2

**Luma3DS version:**

8.0 stable

**Luma3DS configuration/options:**

Default EmuNAND: (1)

Screen brightness: 3

Splash: Before

PIN lock: Off

New 3DS CPU: Clock+L2

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Use custom path: ( )

Enable game patching: (x)

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: (x)

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

--

**Explanation of the issue:**

When I close off Friend List, the screens suddenly go red as soon as I reached HOME Menu, causing to hard-freeze my system, had to force shutdown.
[image]
I did not use "Rosalina" Menu (by pressing L+↓+Select). 
It is not caused by TEST Menu as it's set to HOME Menu, having it set to Test one while on Luma3DS v7.1- and exiting Friend List will cause to trigger TEST Menu instead of Home one.

**Steps to reproduce:**

1. Jump across few homebrew apps (FBI, BootNTR Selector, JK's Save Manager, etc), hitting exit button that instantly kicks me out to HOME Menu, open another homebrew app.

2. Open up Friend List

3. Exit.

**Dump file:**

None.

TuxSH, 2017-06-11:
Looks like a kernel panic, possibly due to a ressource leak.

"Homebrew apps": CIA format or 3DSX?

Maybe the issue is inherent to your particular setup to begin with.

SvenDaHacker64, 2017-06-12:
@TuxSH 
Exactly. I used homebrews as .cia
It also happens after I played a game (Super Mario 3D Land) for a while, enter Friend List then exit.

I've caught some crash dumps while using Luma3DS v8.0-53e2ef24 as recent commit version. It must be a problem with am (Application Manager).
Using TinyFormat for deleting CTRNAND:/data/ kinda seems to help.
[crash_dumps.zip](https://github.com/AuroraWright/Luma3DS/files/1067175/crash_dumps.zip)

TuxSH, 2017-06-13:
[boot.zip](https://github.com/AuroraWright/Luma3DS/files/1070024/boot.zip)
What about the latest commit above ^ ?

SvenDaHacker64, 2017-06-13:
I think it's running fine now after I tried your boot.firm, thanks. I'll let you know if this happens again.

closed 2017-06-13 by SvenDaHacker64

SvenDaHacker64, 2017-06-13:
@TuxSH 
Ugh... I got backfired, it crashes again. I was playing Harmoknight, closed that game, did same thing with Friend List.