LumaTeam/Luma3DS #593: Clock+L2 defaults to On in Rosalina, even when turned off in Luma config

gmonteir, 2017-06-11:
**System model:**

[e.g. 2DS, New 3DS, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**

[e.g. 11.4.0-37U SysNAND, 11.4.0-37J EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**

[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**

[e.g. 7.1 stable or if using nightly/hourly specify the commit like this https://github.com/AuroraWright/Luma3DS/commit/9570e6cbeca53128433abbf5e3473cb8a07fe69e]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Use custom path: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

--

**Explanation of the issue:**

**Steps to reproduce:**

1.

2.

**Dump file:**

Every time I boot, I have to turn off Clock+L2 manually, because Rosalina seems to ignore the Luma config option.

Clement95, 2017-06-11:
> 9570e6cbeca53128433abbf5e3473cb8a07fe69e]

If you want to turn off cpu clock, turn off it in the Luma config

bakugo, 2017-06-11:
The N3DS menu in rosalina doesn't update when the clock changes (when you have Clock+L2 disabled, the clock changes to O3DS values when you start a game that doesn't use N3DS values) but don't worry, it's still disabled.

That said, it really should be changed to update every time the menu is accessed.

Hikari-chin, 2017-06-12:
The Rosalina menu currently doesn't update to reflect changes in clock rate/L2 cache; they display as on because the Home Menu runs in Clock+L2 mode stock/by default on N3DS. Despite this, the values are actually being changed normally when opening an non-N3DS enhanced title, provided you haven't forced them with the config menu option.

closed 2017-06-12 by TuxSH

Hikari-chin, 2017-06-12:
This has been fixed in the latest commit. Thanks for the report.

commit 2ff97185 (2017-06-12): Merge pull request #595 from Hikari-chin/master

Fix #593
commit 33af9cfe (2017-06-12): Fix #593