LumaTeam/Luma3DS #598: Crash with Stack Dump when launching games with NTR Mode3

InkDragon, 2017-06-12:
**System model:** o3ds
**SysNAND version:** sys 11.4.0-37u
**Entrypoint:** B9S 1.2
**Luma3DS version:** 8.0
**Luma3DS configuration/options:**

Default EmuNAND: (1)
Screen brightness: (4)
Splash: (off)
PIN lock: (off)

--

Autoboot EmuNAND: ( )
Use EmuNAND FIRM if booting with R: ( )
Enable loading external FIRMs and modules: ( )
Use custom path: ( )
Enable game patching: (X)
Show NAND or user string in System Settings: (X)
Show GBA boot screen in patched AGB_FIRM: ( )
Patch ARM9 access: ( )
Set developer UNITINFO: ( )

--

**Explanation of the issue:**
System crashes with a stack dump error when booting MHG via mode 3 NTR boot Selector

**Steps to reproduce:**

1. load bootNTR selector mode 3, lode NTR more 3 version 3.4
2.  press home and load Monster Hunter Generations
3. Game will play as normal first run, but many codes will not work.
4. Save & quit game, reboot system
5. Repeat steps 1 & 2
6. Game will immediately crash and produce an arm 11 error

**Dump file:**
I was unable to get the file to upload via github as a zip, so this is the link to the upload from [my posts in the MHG cheat file thread](https://gbatemp.net/threads/monster-hunter-generation-cheats-plugin.434897/page-17#post-7386886).
https://gbatemp.net/attachments/crash_dump_00000000-rar.89868/

TuxSH, 2017-06-13:
Not sure if it's a Luma3DS, and Luma3DS 8.0 issue.

InkDragon, 2017-06-13:
I attempted to use the luma 8 legacy build which just resulted in the game not loading so likely a luma issue

T3CHNOLOG1C, 2017-06-13:
Attempting to launch BootNTRSelector Mode 3 on my 11.4.0-37U O3DS with latest commit (self built) just softlocks the system.

TuxSH, 2017-06-14:
Should be fixed by the latest commits.
[boot.zip](https://github.com/AuroraWright/Luma3DS/files/1073197/boot.zip)

closed 2017-06-14 by TuxSH

InkDragon, 2017-06-14:
Nope still borked, here's the dump for it.
https://www.dropbox.com/s/5lotkhkq1xsi020/crash_dump_00000001.zip?dl=0

InkDragon, 2017-06-14:
can someone please reopen this as its not fixed

Skullfun, 2017-08-24:
Hi, did you solve your problem?

InkDragon, 2017-08-25:
yes, was fixed. switched to a nightly build and deleted and redownloaded the cheat file.

Skullfun, 2017-08-25:
Where do you download the nightly build? O what is that?

Skullfun, 2017-08-25:
I have the problem with mhxx, I patched language with Luma: Enable game patching: (X)