LumaTeam/Luma3DS #599: [Bug Report] Screen capture broken

RS55, 2017-06-12:
**System model:**

[Old 3DS]

**SysNAND version:**

11.4.0-37U SysNAND

**Entrypoint:**

Boot9Strap/Sighax

**Luma3DS version:**

Luma 3DS v8.0-901d499

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( 4 )

Splash: ( )

PIN lock: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Use custom path: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: ( X )

Show GBA boot screen in patched AGB_FIRM: ( X )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

--

I have been using the screenshot function and after taking an X quantity, this occurs... Or sometimes the first with several games.

PD: This happened to me with the stable Luma3DS v8.0 and now with Luma 3DS v8.0-901d499 (And other nightlies I've used)

[image]

TuxSH, 2017-06-13:
Happens occasionnaly, will fix later.

closed 2017-06-13 by TuxSH

Hikari-chin, 2017-06-25:
This should be fixed in commit 2363817.