LumaTeam/Luma3DS #600: Old Luma updating to new Luma without chainloader

Japoelho, 2017-06-13:
**System model:** Old 3DS
**SysNAND version (+emu/redNAND version if applicable):** 11.2.0-35U SysNAND
**Entrypoint (How/what you're using to boot Luma3DS):** arm9loaderhax
**Luma3DS version:** 6.6-b16d887 (dev)
**Luma3DS configuration/options:**
Default EmuNAND: ( )
Screen brightness: ( )
Splash: ( )
PIN lock: ( )
New 3DS CPU: ( )
Autoboot EmuNAND: ( )
Use EmuNAND FIRM if booting with R: ( )
Enable loading external FIRMs and modules: ( )
Use custom path: ( )
Enable game patching: ( )
Show NAND or user string in System Settings: ( x )
Show GBA boot screen in patched AGB_FIRM: ( )
Patch ARM9 access: ( )
Set developer UNITINFO: ( )

**Explanation of the issue:**
I have an old Luma version, 6.6, and I tried updating to B9S. This version doesn't have the chainloader menu and I can't update to the latest Luma using the Updater because of the B9S requirement.
How can I update the Luma manually on the NAND or how can I make the Luma Updater update to a newer but still A9LH compatible first and then to B9S?

**Steps to reproduce:**
1. Luma 6.6 installed
2. Try to update it.

**Dump file:**

Hikari-chin, 2017-06-13:
Follow https://3ds.guide/a9lh-to-b9s then https://3ds.guide/updating-b9s.

closed 2017-06-13 by Hikari-chin

Japoelho, 2017-06-13:
I'm searching for the how-to do the manual update but I can't find it, do you know where can I find it or explain it if possible?

Hikari-chin, 2017-06-13:
Extract the arm9loaderhax.bin from the linked .7z file and replace the arm9loaderhax.bin on your SD card with it.

Japoelho, 2017-06-13:
Thank you very much.