LumaTeam/Luma3DS #602: (Reproducable) An exception occurred- Data Abort 3dsx_app

An-attempted-inevitability, 2017-06-13:
**System model:**
 [Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**

[e.g. 11.4.0-37U SysNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap

**Luma3DS version:**

[Luma3ds v8,0]

Checked through Rosalina menu (see explanation below)

**Luma3DS configuration/options:**

Default EmuNAND: ( ) Not on Configuration Screen?

Screen brightness: (4)

Splash: (off)

PIN lock: (off)

New 3DS CPU: (n/a)

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Use custom path: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

--

**Explanation of the issue:**
This exception comes up whenever opening FBI from homebrew screen (L+DOWN+SELECT rosalina method) preventing completion of finalization process for Luma3ds.

Also worth noting that I am trying Homebrew for the first time and signed up for github account for advice or a solution. 
If I am in the wrong place to look for answers, a kind point in the right direction would be greatly appreciated.

Thanks.

**Steps to reproduce:**

1. Launch the Download Play application

2. Open Rosalina menu (L+Down+Select)

3. Select Miscellaneous options

4, Select Switch the hb. title to the current app.

5. Press B Three times to Exit

6. Exit Download Play to Home Screen

7. Launch the Download Play application

8.Open FBI from Homebrew Launcher

9.. Exception screen will pop up

**Dump file:**
[crash_dump_00000000.zip](https://github.com/AuroraWright/Luma3DS/files/1070196/crash_dump_00000000.zip)

Margen67, 2017-06-13:
I can use FBI 3dsx from Rosalina without any problems.

Does this occur with only Download Play?

An-attempted-inevitability, 2017-06-13:
I am new to this and didn't think you could do other apps. Just tried with two other ones Activity Log and Face Raiders and had same data abort issue.

An-attempted-inevitability, 2017-06-13:
I forgot to mention that I was following the guide on https://3ds.guide/ for old 3ds 8.1  and am stuck on the launch FBI step because of the error.

T3CHNOLOG1C, 2017-06-13:
If needed, I can try making an emunand and downgrading it to 8.1 to see if I can reproduce the problem.

PerrinLigh, 2017-06-13:
I do have the same problem but I upgraded my Old 3DS from 11.2, I can't start anything on the Homebrew Launcher it crashes everytimes

fincs, 2017-06-13:
In theory this should be fixed by 005684d86511797d68254d2586cc8ce075bc6629

PerrinLigh, 2017-06-13:
Thanks man ! And how do I install this ? (I'm a rookie with code/hack)

FerdinandoPH, 2017-06-13:
Hi.
I have the same problem of PerrinLigh. I have a 2ds that was of 11.0 (now 11.4) and I was doing the setup and it failed because the same problem.
Please, can tou create the .firm file? I don´t have idea to use that code!!!

TuxSH, 2017-06-14:
[boot.zip](https://github.com/AuroraWright/Luma3DS/files/1073202/boot.zip)

Shadow322, 2017-06-14:
so this is commit a1b2ea2c5bdad2b91d5d67d84ebb1b5481bf6131 ?

FerdinandoPH, 2017-06-14:
Thanks!
One thing more, where I should place it? I put it in root and continue failing

chrisehm, 2017-06-14:
[image]

This happens after I try to launch FBI after I open homebrew via download play. Can anyone help?

FerdinandoPH, 2017-06-14:
It´s the sames that it happened to me

FerdinandoPH, 2017-06-14:
Please, how can I do?

Margen67, 2017-06-14:
@FerdinandoPH http://bfy.tw/4kQb

QuirkyRose, 2017-06-14:
im also having the same issue and dont know what to do

An-attempted-inevitability, 2017-06-15:
So I had tried to drop the new compiled boot.zip from @TuxSH (Since I don't know codes either... thanks btw)  into the root of the card replacing the old boot file and tried to run FBI again and got to this exception. 

[image]

[crash_dump_00000000.zip](https://github.com/AuroraWright/Luma3DS/files/1076332/crash_dump_00000000.zip)

I glad to see so many people helping out and hope we can find solutions to all these issues.
Thanks everyone

FerdinandoPH, 2017-06-15:
I did the same, and it fails to me too

tomk2535, 2017-06-15:
Any solution yet? Keep getting this excact same crash.

Particularly annoying, as I had a working gateway-hack, but decided to change to Luma3DS. Already managed to perform this hack on a new 3ds, but when doing it on my own old 3ds things does not go that flawlessly...

swatgoss, 2017-06-15:
Hi,
I have an Old3DS that was updated to 11.3 before following 3DSGuide.

I'm also stuck at the step where i'm supposed to execute FBI ...
It appears that i can't successfully run any other application displayed in the homebrew menu !

I got multiple crash dumps that i join here in hope that someone will get it right.
[crash_dump_0000000X.zip](https://github.com/AuroraWright/Luma3DS/files/1078494/crash_dump_0000000X.zip)

QuirkyRose, 2017-06-15:
dont count me on this but iv read the latest vertion of luma has issues, im going to try a legacy builld and see what results i get

swatgoss, 2017-06-15:
You need a *hax that allows you to launch code at kernel level (backdoor thingy).

What is the status of your 3DS right now ?
What version of Firmware are you running ?
What version of Luma and HomeBrew Menu are you running ?

All those infos will really help to assist if you have a more specific problem than trying to run "mismatched and not up to date versions of programs/hax/payloads"

QuirkyRose, 2017-06-15:
sorry but i dont know much about how this works but i downloaded the lastest vertions of everything, but iv seen multiple posts with this same issue all dating no further back than the 6th of this month and appantly is an issue with luma and its needs to update?

swatgoss, 2017-06-15:
I hacked my Old3DS (red) only today but i read that the last 11.3 kernel exploits by smealum were only released quite recently (february maybe ?)
Those new "hacks" and the 11.4 Official Nintendo firmware update made a big effect on the scene, so many guides and tutorials are just PURE garbage now !

If you want fast and maybe efficient help, please at least post your infos : 3DS firmware version, already done hacks, version of the programs that are failing to run.

If you can take note of those versions and try newer ones and also note the results that you have, it would be really usefull for everybody, you included !

QuirkyRose, 2017-06-15:
ill try
luma v 8.0
GodMode9-20170531-232521.zip
3ds :sys 11.4.0-37E
homebrew launcher says:1.1.1
iv run safehax and udsploit
all done with boot9strap
originaly hacked with ninja hax 

is there anyother info you need?

QuirkyRose, 2017-06-15:
and if its needed (dont know why it would be) the serial code is EF10700262[4] and its a shiny red old 3ds

chrisehm, 2017-06-15:
Thank you @swatgoss the boot.3dsx file helped FBI launch like a charm

swatgoss, 2017-06-15:
Really try to replace the boot.3dsx file at the root of the SDCARD with the latest version of HomeBrewMenu

https://github.com/fincs/new-hbmenu/releases

Thats what did the trick for my Old3DS ;)

Also, the page https://github.com/AuroraWright/Luma3DS/releases/tag/v8.0
CLEARLY STATES that you need AT LEAST homebrew menu 2.0 (3dsx loading section)

Keep a backup of you actual files and try the latest ones and let know if this change your situation ;)

QuirkyRose, 2017-06-15:
thanks @swatgoss sorry for being stupid :p updating homebrew worked

swatgoss, 2017-06-15:
I know its hard to understand so much "technical information" at once, but PLEASE read the documentations/release notes that the developpers took time to put together, they already gave you all the answers that you crave to use the step by step brainless guides that are out there everywhere ...

Thats really not a good idea to download/run/inject/modify software on any machine you care about by following even the best looking guides on a website.
Think for yourself and take care of your console as if it was your own body ... dont put things found on the curb before activating your brain at least a little bit !

closed 2017-06-15 by fincs

Yelley09, 2021-05-30:
I know it already passed a few years, but i keep having the same problem