LumaTeam/Luma3DS #603: "An Exception Occurred" error when loading "Shaun the Sheep 1-15 EUR" on US o3DSxl

rynho, 2017-06-14:
**System model:**
Old 3DS XL

**SysNAND version (+emu/redNAND version if applicable):**
11.4.0-37U SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**
Boot9Strap 1.2

**Luma3DS version:**
8.0 stable

**Luma3DS configuration/options:**
Default EmuNAND: ( 1)
Screen brightness: ( 1)
Splash: ( off)
PIN lock: ( off)
New 3DS CPU: ( )
--
Autoboot EmuNAND: ( no )
Use EmuNAND FIRM if booting with R: ( no )
Enable loading external FIRMs and modules: ( no )
Enable game patching: ( no )
Show NAND or user string in System Settings: ( yes )
Show GBA boot screen in patched AGB_FIRM: ( no )
Patch ARM9 access: ( no )
Set developer UNITINFO: ( no )
--
**Explanation of the issue:**
downloaded "shaun the sheep (episode 1-15 + Making of)" using CIAngel, "CTR-N-MBMP", EUR, 00040000000f2a00, and install using FBI.  after successful installation, launch the title, and get an ARM11 exception error.  I attached the dump file below.

**Steps to reproduce:**
1. Launch latest CIAngel, choose "Create CIA", region "off", search for "shaun", and select "Shaun the sheep (episode 1-15 _ (EUR) CTR-N-MBMP"

2. After download, exit CIAngel, launch latest FBI and install title from "SD\CIAngel" folder.

3. exit FBI, launch the new installed app, and get the error and found dump in "SD\luma\dumps\arm11"

**Dump file:**

[crash_dump_00000000.zip](https://github.com/AuroraWright/Luma3DS/files/1073549/crash_dump_00000000.zip)

MelonGx, 2017-06-15:
Why don't you write a Region Patch before asking this question?

rynho, 2017-06-15:
I tried, seems didn't work.  Not sure if I didn't use it correctly, since I have some space at the end of the locale.txt.  Also, I read an reply on [this post](https://gbatemp.net/threads/luma3ds-noob-proof-3ds-custom-firmware.411110/page-406&usg=AFQjCNHKfYZC06JXADnfOzCkeG6qMtEmyA&sig2=pX56E5RARjmqPmesieWZkg&cad=rja), which mentioned this title should be work fine on previous Luma3DS version.

I attached the screenshot of that reply on Google result page below, as I don't have access to GBATEMP right now:
[image]

rynho, 2017-06-16:
I used latest nightly build last night, the title running well with enabling the region simulation.  Therefore, I confirm this is a bug regarding region patch in 8.0 stable build, and was fixed in nightly build.  @Hikari-chin could you please close this post, thanks.

closed 2017-06-16 by Hikari-chin