LumaTeam/Luma3DS #604: NSMB2 Luma patching seems to only work on USA versions

Gimzie, 2017-06-14:
A while ago I reported an issue where NSMB2 Luma patching did not work, and then it was fixed. Only, it seems like it was fixed for US versions only. (And possibly only n3DS) There have been many who have tried it with a European version, only for it to not work. I myself was able to test this with a European copy, and it indeed didn't work.

Jan200101, 2017-06-14:
I've tested a mod called New Super Mario Classic on my EU device with a EU Version of the game.
It seems to work fine for me

Gimzie, 2017-06-14:
Is it the Special Edition?

Jan200101, 2017-06-14:
No

GRAnimated, 2017-06-17:
Just tested myself, on the EU version it does work but the special edition does not, but the US regular and special works.

Gimzie, 2017-06-19:
Maybe it's just Special Edition?

Hikari-chin, 2017-06-19:
Are the mods you're testing stated to be compatible with the Special Edition? It's likely there are some small differences that may break certain mods/patches.

Jan200101, 2017-06-19:
@Gimzie Can you try the mod NSMB Classic ?
According to the Devs it should work with the Special Edition for EU

Jan200101, 2017-06-19:
After some searching I noticed
There is no "gold edition" of the EU version instead the normal EU version is the gold edition

Hikari-chin, 2017-06-21:
Closing this; original reporter has not responded and based on @Jan200101's comments it is likely user error.

closed 2017-06-21 by Hikari-chin

AuroraWright, 2017-07-05:
Can't reproduce with both the special edition and a cart, so I confirm this to be user error (or the game files aren't universal)