LumaTeam/Luma3DS #607: Rosalina crashes on 8.1.0-19U

Myriachan, 2017-06-14:
**System model:**

Old 2DS

**SysNAND version (+emu/redNAND version if applicable):**

8.1.0-19U

**Entrypoint (How/what you're using to boot Luma3DS):**

boot9strap

**Luma3DS version:**

8.0

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Use custom path: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

--

**Explanation of the issue:**

I used not-yet-public ntrboothax to directly install boot9strap on a system that has never booted before.  I did this so that I could get a pristine pre-first-boot NAND dump.

**Steps to reproduce:**

1. Have a 3DS on 8.1.0-19x.

2. Attempt to boot the OS from Luma.

**Dump file:**

[crash_dump_00000003.zip](https://github.com/AuroraWright/Luma3DS/files/1074063/crash_dump_00000003.zip)

Hikari-chin, 2017-06-14:
Known issue. Rosalina doesn't current work with firmware <9.0. It's being looked into.

MelonGx, 2017-06-15:
Since 6.5+ Luma crashed or unable to load FIRM on factory stock firmwares.
I think OP's is a same issue.

Valeri0p, 2017-06-15:
_Maybe_ a [legacy bulid](https://astronautlevel2.github.io/Luma3DS/builds/Luma3DS-3a14ee1.zip) can help?

TuxSH, 2017-06-16:
Fixed.

closed 2017-06-16 by TuxSH