LumaTeam/Luma3DS #608: On New3DS, nightly version of Luma auto-shutdown when launching GBA VC game or DSiWare

TheNain38, 2017-06-14:
**System model:**

New 3DS

**SysNAND version (+emu/redNAND version if applicable):**

11.4.0-37E SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap

**Luma3DS version:**

nightly https://github.com/AuroraWright/Luma3DS/commit/24de7c527287ad6bd1b76b9962ef0f4330382acb

**Luma3DS configuration/options:**

Default EmuNAND: (1)

Screen brightness: (1)

Splash: (Before)

PIN lock: (0)

New 3DS CPU: (L2 Cache + Clock)

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Use custom path: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (X)

Show GBA boot screen in patched AGB_FIRM: (X)

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

--

**Explanation of the issue:**

When trying to launch a GBA VC game, or a DSiWare title, the console auto-shutdown, also I don't have this bug on Luma v8.0, and just in case, I even reinstalled a clean TWL_FIRM and AGB_FIRM, by using the instructions on the [wiki](https://github.com/AuroraWright/Luma3DS/wiki/FAQ-and-troubleshooting#i-get-an-auto-shutdown-with-gbadsdsiware-games)

Maybe it's a problem with Luma failing to patch the firms?

**Steps to reproduce:**

1. Try to start a GBA VC game, or a DSiWare title

2. See that the console auto-shutdown

DeathAngel74, 2017-06-14:
Try 005684

DeathAngel74, 2017-06-14:
d8

TuxSH, 2017-06-14:
[boot.zip](https://github.com/AuroraWright/Luma3DS/files/1075441/boot.zip)
Latest commit.

TheNain38, 2017-06-14:
It still doesn't work with the latest commit

TuxSH, 2017-06-14:
I can't reproduce...

TheNain38, 2017-06-14:
The weird thing, is that I don't have this bug on Luma v8.0, and that in between, it looks like you didn't change anything related to TWL_FIRM or AGB_FIRM, or maybe I'm wrong? Because it does as if Luma would fail to apply patching... hmm...

DeathAngel74, 2017-06-15:
Just tested this commit works. 12b561a2.

DeathAngel74, 2017-06-15:
https://drive.google.com/file/d/0B007JgCLgXQLX0d4UHNKZHdhZm8/view?usp=drivesdk

DeathAngel74, 2017-06-15:
Just rename to boot.firm

MelonGx, 2017-06-15:
Haven't got your issue on d030c0b.

DeathAngel74, 2017-06-15:
005684d8 is when the prob started

MelonGx, 2017-06-15:
@DeathAngel74 Then why I have no error on launching DSiWare with da30c0b?

DeathAngel74, 2017-06-15:
Could be any number of variables. I broke 2x 2ds's on 005684d. Pokemon Sun stopped working. 12b561a2 fixed the issue on both 2ds's. 2x n3dsxl's weren't effected though.

MelonGx, 2017-06-15:
@DeathAngel74 OK, I got a1b2ea2 have this issue on O3DS 11.4.0-37J.

TheNain38, 2017-06-15:
@TuxSH Does something in one of these [commits](https://github.com/AuroraWright/Luma3DS/commits/master/patches) between v8.0 and the master, that could have caused this?

MelonGx, 2017-06-15:
@TheNain38 My first test was on the New3DS JPN and no issue on both.

TheNain38, 2017-06-15:
Well, it works fine with Luma v8.0, so I'll try to do regression testing and see at which commit it fails... also, I remebered that some commits had the New3DS unable to even boot, I'll report all of that when I'm done

TuxSH, 2017-06-19:
Fixed.

closed 2017-06-19 by TuxSH

TheNain38, 2017-06-19:
Thanks, it works now, could I know what was the problem or you don't really have the time to explain it?