LumaTeam/Luma3DS #610: Crash while loading LayeredFS with Pokémon X

SwampyGator, 2017-06-14:
**System model:**

New 3DS XL

**SysNAND version (+emu/redNAND version if applicable):**

11.4.0-37U (I think, it's whatever the 11.4 is)

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap 1.2

**Luma3DS version:**

Luma3DS 8.0

**Luma3DS configuration/options:**

Default EmuNAND: (1x )

Screen brightness: (4)

Splash: (Before )

PIN lock: (Off)

New 3DS CPU: (Off)

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Use custom path: ( )

Enable game patching: (X)

Show NAND or user string in System Settings: (X)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

--

**Explanation of the issue:**

First I tried to load Pokémon X with an Extreme Randomizer. I just edited it in pk3ds, and copied everything. When it crashed, I assumed I put too many files. So, instead I downloaded External X Insane Mode. Same crash, so I knew it was LayeredFS not my randomizer. Below I have attached my crash log from Eternal X

**Steps to reproduce:**

1. Download Eternal X, and install it to LayeredFS (luma/titles/titleid/romfs)

2. Boot Pokémon X

**Dump file:**

[crash_dump_00000008.zip](https://github.com/AuroraWright/Luma3DS/files/1075368/crash_dump_00000008.zip)

Hikari-chin, 2017-06-14:
Did you test if these modifications work with non-Luma LayeredFS implementations/works as a modified CIA/etc? Also, did you make sure you have the correct update for the game? Some mods only work with the original game, specific updates, etc.

SwampyGator, 2017-06-14:
I'm next to 100% positive that Eternal X has been used by more than just me. And as it states it runs on base game, I've uninstalled my game update.

Hikari-chin, 2017-06-14:
That didn't answer the question. Please try your modifications with NTR's LayeredFS plugin, OnionFS, or something of that nature and confirm that it works or doesn't work there.

AuroraWright, 2017-06-14:
Can you check if it crashes or not if you just have an empty "romfs" folder?

SwampyGator, 2017-06-14:
@Hikari-chin OnionFS crashes.

@AuroraWright Empty RomFS folder also crashes.

TuxSH, 2017-06-14:
Can't reproduce with an empty folder.

closed 2017-06-14 by TuxSH

SwampyGator, 2017-06-14:
I'm not sure how, but somehow it didn't uninstall despite me deleting it. Sorry everyone!