LumaTeam/Luma3DS #614: Problem while entering to the System Settings app (An exception occurred blackscreen)

Constoush, 2017-06-15:
**System model:**

[e.g. 2DS]

**SysNAND version (+emu/redNAND version if applicable):**

[e.g. 11.4.0-37EUR SysNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**

[Boot9Strap]

**Luma3DS version:**

[8.0]

**Luma3DS configuration/options:**

Default EmuNAND: (1)

Screen brightness: (4)

Splash: (Off)

PIN lock: (Off)
--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (X)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

--

**Explanation of the issue:**
After Updating boot9strap to 1.2 and luma to 8.0 with the guide on 3ds.guide on my first SD card, I can't enter into the system settings anymore, this black screen pop up
[image]
[image]
I have the files updated on another SD card and this black screen doesn't pop up, such as without SD card it doesn't pop up too.
I tried to move the exact same files from the working SD card to the non-working SD card and it doesn't work.
I don't know why this happens and why one one SD card and not on the other, it doesn't make sense.
I don't know if somebody else has the same issue, I saw something like that on another thread but the solution of this problem doesn't work for me.. I don't know if there's something else to do..
On the SD card that doesn't work I installed fake badges, maybe that's why it doesn't work anymore, but I don't know which folder to delete to let my 2DS create new Home Menu Files, and entering into System Settings through the regular Ninento App or the Badge seems to do the same.

**Steps to reproduce:**

1. Just enter into the system settings

2. There's the error.

**Dump file:**
I can't upload my crash dump, even in ZIP format for some reason

Constoush, 2017-06-15:
[https://drive.google.com/open?id=0BwNj-VtfRxIOaUx5dzI1alA1azg](url) here's the link to the dump file, I can't upload it directly on the website for some reason

Constoush, 2017-06-15:
With some tests I found that entering into the system settings through the Home Menu options works, it's definitely a problem with the disposition of the home menu, it might be an incompatibility with the MrCheeze's app GYTB if it could help someone who's encountering the same issue

ZenIsBestWolf, 2017-06-15:
While I do not possess a EUR 2ds, on my o3ds xl USA I cannot reproduce. Works fine for me.

_Sent from my BLU Life One X2 using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

Constoush, 2017-06-15:
I think it was a problem with my Home Menu data, so it can't be reproduced on a console that works fine

closed 2017-06-15 by AuroraWright