LumaTeam/Luma3DS #615: Unable to start homebrew with RomFS redirection

HoLLy-HaCKeR, 2017-06-15:
**System model:**

o3DS

**SysNAND version (+emu/redNAND version if applicable):**

11.4.0-37E SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

B9S

**Luma3DS version:**

8.0

**Luma3DS configuration/options:**

*-removed disabled/unused options-*

Enable loading external FIRMs and modules: ( )

Use custom path: ( )

Enable game patching: (**X**)

Show NAND or user string in System Settings: (**X**)

Set developer UNITINFO: ( )

--

**Explanation of the issue:**

When launching a homebrew title for which custom romfs is set in luma/titles/[id], the app will crash on startup. The romfs folder does not have to contain anything, and this also breaks apps that do not even have romfs.

Tested on the following:
* 3DSident (00040000 01610000)
* PKSM (00040000 0EC10000)
* haxagon (00040000 0099AA00), see crash dump 13
* minicraft 3ds (00040000 0FF4FF00, has no romfs), see crash dump 14

**Steps to reproduce:**

1. Create the folder luma/titles/0004000001610000/romfs.

2. Try to start 3DSident.

**Dump file:**

[crash_dump_00000013.zip](https://github.com/AuroraWright/Luma3DS/files/1077592/crash_dump_00000013.zip)
[crash_dump_00000014.zip](https://github.com/AuroraWright/Luma3DS/files/1077593/crash_dump_00000014.zip)

HoLLy-HaCKeR, 2017-06-15:
Found this while trying to create a skin for 3DSident. Normal games work fine with custom RomFS (tested with Maple Story, 00040000 00111600). 

Wild guess would be high title id's, since they are all above 00800000.

AuroraWright, 2017-06-15:
LayeredFS uses code symbols only present in official SDK apps, so it won't work with homebrew

closed 2017-06-15 by AuroraWright

HoLLy-HaCKeR, 2017-06-15:
No plans to make it work with DevkitPro? I'd think you'll only need to add a check to see what library/SDK was used for the app and hook a different method depending on that.

TuxSH, 2017-06-15:
Not doable. The reasons why the current approach has been chosen is update data's romfs.