LumaTeam/Luma3DS #617: Black screen on wakeup

LemonJuiz, 2017-06-15:
**System model:**
New 3DS

**SysNAND version (+emu/redNAND version if applicable):**
11.4.0-37U SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**
Boot9Strap/Sighax

**Luma3DS version:**
8.0 stable

**Luma3DS configuration/options:**
Default EmuNAND: (1)

Screen brightness: (4)

Splash: (Off)

PIN lock: (Off)

New 3DS CPU: (Off)

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Use custom path: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (X)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

--

**Explanation of the issue:**
When closing the N3DS, after waiting about 4 minutes, upon reopening the device the screen is completely black and you have to force shutdown to get it to work again. Trying this Github first and if it turns out that this is a B9S problem, I'll post an issue there.

**Steps to reproduce:**

1. Close device

2. Wait

sora10pls, 2017-06-15:
Unable to replicate.

B9S 1.2, Luma3DS v8.0, New 3DS, 11.4.0-37J.

Derek1199, 2017-06-15:
This is also an issue I have never experienced. This has happened to me when I was on stock firmware though before, but it has nothing to do with firmware. It only happened to me when my 3ds got too cold.

LemonJuiz, 2017-06-15:
I've tried different ways to replicate but issue seems to not occur anymore. It happened 4-5 times in one day and twice the day after. Might've been a very select issue. Closing.

closed 2017-06-15 by LemonJuiz

pcs3rd, 2017-06-20:
I have this happen sometimes (most commonly after leaving it closed after accidentally dropping it.)