LumaTeam/Luma3DS #618: Feature Request : Prevent 3DS from disconnecting from WiFi if no internet is present

deadair3210, 2017-06-15:
**System model:**

[e.g. 2DS, New 3DS, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**
11.4.0-37
[e.g. 11.4.0-37U SysNAND, 11.4.0-37J EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**
Boot9Strap
[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**
8.0
[e.g. 7.1 stable or if using nightly/hourly specify the commit like this https://github.com/AuroraWright/Luma3DS/commit/9570e6cbeca53128433abbf5e3473cb8a07fe69e]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Use custom path: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

--

**Explanation of the issue:**

Since we now have the ability to control the entire system, would it be possible to keep the 3DS from disconnecting from WiFi when it can't talk to nintendo

**Steps to reproduce:**

1.

2.

**Dump file:**

Hikari-chin, 2017-06-15:
So, despite the very...interesting choice of title, are you trying to ask if it's possible to bypass the connection to Nintendo's servers during the connection test?

piratesephiroth, 2017-06-15:
I'm pretty sure this has been requested before a long time ago.

Yeah, normally a successful connection test is required for using the wireless connection.
MicroSD Management is the only app capable of bypassing that.

Hikari-chin, 2017-06-16:
Unfortunately, this is infeasible to implement. The 3DS uses that connection test to choose which wireless network to connect to. mSD Management only "bypasses" it because it forces you to manually select a network.

closed 2017-06-16 by Hikari-chin

TuxSH, 2017-06-16:
So yeah, µSD management doesn't really bypass it, it simply isn't affected.

Tbh I'd have implemented it if it were feasible.

deadair3210, 2017-06-16:
Thanks anyway, figured it was worth a shot