LumaTeam/Luma3DS #619: [Compilation] Current and future features for Rosalina (All Rosalina feature requests go here!)

Hikari-chin, 2017-06-16:
With the release of Luma3DS 8.0 and the custom sysmodule Rosalina, we have been getting a lot of issues concerning Rosalina's current/planned features, as well as requests for new features. We've created this issue to consolidate these requests; **if you want to request a feature for Rosalina, please do so here**. If a feature has already been requested, please do not comment about your support or dislike of said feature; instead, use Github's "reaction" system to give it a thumbs up, thumbs down, like, etc. Please note that any arguing, bickering, and/or drama will be removed without notice.

## Original Features
These were Rosalina's original features; i.e. the features available in the v8.0 release of Luma3DS.
* **Battery Percentage:** The battery percentage is displayed in the lower right corner of the Rosalina menu.
* **Exception Handlers:** Advanced Exception Handlers for catching errors, crashes, etc.
* **GDB Stub:** Fully featured GDB stub for debugging purposes, compatible with GDB Server, IDA, etc.
* **Homebrew Menu takeover title:** The title that is taken over by the Homebrew Menu on launch can be changed.
* **InputRedirection:** For using an external controller to control your 3DS.
* **New 3DS Clock/L2:** The clock rate and L2 cache can be adjusted in real time to run in Old 3DS or New 3DS mode, **New 3DS only**
* **Process List:** List of currently running processes.
* **Process Patches:** SM service checks and FS archive checks can be patched out.
* **Rosalina menu:** Interactive menu with various options, triggered by L+Down+Select. The default combo is changeable.
* **Screenshots:** Saves the top and bottom screen as BMP images.

## Features implemented at the request of the community
These are features that were requested by the community and have been successfully implemented.
* **[Power off/Reboot](https://github.com/AuroraWright/Luma3DS/issues/575):** Options to power off or reboot the system are now available in the Rosalina menu. Implemented in commit 37eb21d.
* **[Key combo behavior change](https://github.com/AuroraWright/Luma3DS/issues/491):** Rosalina waits for all keys to be released before setting the trigger combo, making it much easier to set correctly. Implemented in commit 60d136f.
* **[Optional ARM11 Exception Handlers](https://github.com/AuroraWright/Luma3DS/pull/561):** The ability to disable the Exception Handlers was requested several times, was implemented in a PR by SciresM, and was accepted/merged in commit 4d99143.
* **[Display local IP address when InputRedirection is on](https://github.com/AuroraWright/Luma3DS/issues/613):** Rosalina now displays the local IP address when InputRedirection is turned on, for ease of access. Originally, it only displayed when the GDB stub was active. This was implemented in commit 9ccfacd.

## Features accepted/planned to be implemented
These are features that have been requested by the community and have been accepted; they will be implemented at some point in the future.
* **[Brightness adjustment in the Rosalina menu](https://github.com/AuroraWright/Luma3DS/issues/571):** This would be convenient to be able to be adjusted on the fly, instead of needing to return to the Home Menu to do so. This has been requested a few times and is being worked on.
* **[Persistence of custom choices](https://github.com/AuroraWright/Luma3DS/issues/489)** - Currently, Rosalina doesn't save custom key combos, [choice of Homebrew Menu takeover title](https://github.com/AuroraWright/Luma3DS/issues/575), etc. across reboots. This has been requested a few times and is being worked on.

## Features under consideration
These are features that have been requested by the community and are currently being considered as possible candidates for future implementation.
* **[3D strength adjustment in the Rosalina menu](https://github.com/AuroraWright/Luma3DS/issues/597):** This would be useful for people with broken 3D sliders.
* **[NTR-esque plugin system](https://github.com/AuroraWright/Luma3DS/issues/512):** This has been requested a bunch, and is being considered.
* **[Volume adjustment in the Rosalina menu](https://github.com/AuroraWright/Luma3DS/issues/576):** This would be useful for people with broken volume sliders.
* **[WiFi connection switcher](https://github.com/AuroraWright/Luma3DS/issues/520):** The ability to select which WiFi network to connect to from the three available saved networks, from the Rosalina menu. 

## Rejected features
These are features requested by the community that were not appropriate for Rosalina, leading them to be rejected and not implemented.
* **[Make Rosalina optional](https://github.com/AuroraWright/Luma3DS/issues/496):** This feature was rejected because several key features and programs now rely on Rosalina to function properly, namely, the [Homebrew Menu](https://github.com/fincs/new-hbmenu).
* **[Screenshot output to non-BMP image formats](https://github.com/AuroraWright/Luma3DS/issues/573):** This feature was rejected because converting BMP to PNG, JPG, WEBP, etc. is already very easy to do outside of Rosalina, and adding the feature would make an already-slow process even slower.

Plailect, 2017-06-16:
An integrated FTP/SFTP server with access to SDMC, CTRNAND, and TWLNAND would be an excellent addition to Rosalina.

walterclozet, 2017-06-16:
i would like an option to set region stuff for the currently selected title, instead of installing a luma locale switch tool and have to select one title from a large list to set the region info.

Jimmy-Z, 2017-06-16:
I would like a default locale.txt, including multiple lines like this:
````
US EN USA 01
JP JP JPN 01
EU EN GB 01
````
when out of region title launches, select the respective setting based on their region if per title locale.txt doesn't exist.

Valeri0p, 2017-06-16:
It would be practical to implement a Samba server, rather than an FTP one?
(To get rid of mSD management...)
Also, if I can ask, why the arm11 exception handlers switch was disabled?

ZenIsBestWolf, 2017-06-16:
Possibly improve the screenshot tool. HB used to have a screenshot feature upon pushing the home button or some other kind of key combo. This process was fast, and then to get the screenshots, you launch a separate program that puts them into image files. This was useful when using HANS.

Joshkinz, 2017-06-16:
Displaying the IP address somewhere in Rosalina (possibly in the confirmation screen of InputRedirect) would be great, and helpful for using InputRedirect.

WubTheGame, 2017-06-16:
maybe after selecting take screenshot from the rosalina menu, you can have the option to press a button to cancel it before it completes?

Gnarmagon, 2017-06-16:
The Plugin request talks about the Ntr Plugin loading but what about doing some Linda plugin loading like Henkaku does ?

Hikari-chin, 2017-06-16:
@gnmmarechal You're going to have to clarify; HENkaku is an exploit for the Vita, an entirely different system running an entirely different OS, so please specify.

StateOfException, 2017-06-17:
Given the fact that Rosalina allows you to set N3DS Clockspeed/L2 after booting, does this mean that it's now viable to set up per-title configurations for titles with issues with N3DS Clock+L2 that bump them down to O3DS settings whenever they're started/active?

Games like Sega All-Stars Racing Transformed and the SMT: Devil Survivor games are only really consistently stable/playable at O3DS settings. Up until this point, running them has required powering off the system, manually opening Luma's settings panel, and switching the Clock+L2 boost off every single time the user wants to run them, then rebooting and switching the N3DS setting back on after each play session.

The ideal behaviour would be for Luma/Rosalina to detect when an O3DS-only game is being launched (either through matching its ID against a user-provided blacklist in a config file, or through the user setting it to O3DS-mode in Rosalina at runtime, with the setting being remembered for later instances of that title being started), switch off the Clock+L2 boost for the duration of its runtime, then flip back to N3DS settings when the game has finished running.

Is that kind of behaviour possible now? Whenever it's been raised before, it was dismissed as being fundamentally impossible with the way Luma handles the Clock+L2 boost- has that changed?

HoLLy-HaCKeR, 2017-06-17:
@Gnarmagon the plugin request talks about plugins in general, they wouldn't have to be in the same format as NTR plugins.

Gnarmagon, 2017-06-17:
@Hikari-chin @HoLLy-HaCKeR I just thought it would be cool if you could develop your own Plugins that run natively on Rosalina without the need of starting a Homebrew or  building a Forwarder Payload...
Here are a bit mlre Infos on what it does: https://tai.henkaku.xyz

HoLLy-HaCKeR, 2017-06-17:
@Gnarmagon that's exactly what the original issue was about though, have you read it completely?

Hikari-chin, 2017-06-17:
@StateOfException Currently, you can manually go into the N3DS menu and disable/enable the clock/L2 boost in real time, yes. Luma doesn't currently do what you stated, but it is likely feasible. Thanks for the request.

RS55, 2017-06-17:
It is possible to add an option that can be activated and deactivated, to disable the Home button and toggle it as a shortcut to take screenshots? Something like what happens with HANS but that can be changed in rosalina.

Derek1199, 2017-06-17:
Having a dump codebins and load codebins feature on luma would be nice. I'd also like a force test menu NS patch. Lol.

WubTheGame, 2017-06-17:
when exiting the Rosalina menu, any chance the B button can stop being active after closing the menu? Example of how it currently is: watching Youtube, open Rosalina menu, Exit, Youtube goes back a page.

ahoischen, 2017-06-17:
I think blocking other software from reading inputs for a certain amount of time would be very hard to implement, but it would be feasible to wait for maybe half a second before closing Rosalina and resuming the application. It would, of course, be best if this delay was optional.

Ennea, 2017-06-17:
WiFi toggle would be nice. It's a hassle on N3DS when you have to go back to the home menu.

Uumas, 2017-06-18:
A Luma config option to disable the menu would be nice.

Hikari-chin, 2017-06-18:
@Uumas See "rejected features", it mentions this. Rosalina will not be made optional.

Margen67, 2017-06-18:
https://github.com/AuroraWright/Luma3DS/commit/17828273a5a9450658fc36555cf7892ded7e65b2

maorninja, 2017-06-19:
Wait, it now logs when people delete other peoples text? COOOL!

maorninja, 2017-06-19:
Is it possible to make it use payloads again?

Hikari-chin, 2017-06-19:
@maorninja You're going to have to clarify. What do you mean, "make it use payloads again"? Luma is still capable of chainloading ARM9 payloads. If you mean .3dsx *hax payloads, then no, it isn't possible; the new Homebrew Menu was not developed with *hax payloads in mind.

maorninja, 2017-06-19:
Yeah, I meant the .3dsx

Also, I mean't by if you can use it by Luma's side, not actually open it.
And yes, it was made with *hax payloads in mind.

fincs, 2017-06-19:
> the new Homebrew Menu was not developed with *hax payloads in mind.

Not true. hbmenu 2.0 still supports *hax 2.x payloads and even ninjhax 1.x.

Hikari-chin, 2017-06-19:
@maorninja The new Homebrew Menu already works with .3dsx files still. But you're going to have to clarify: are you wanting Luma to be able to open .3dsx files directly, without the Homebrew Menu?

maorninja, 2017-06-19:
No, as in, will hax payloads still work on Luma 8.0?

maorninja, 2017-06-19:
@fincs I wonder which exploits use 1.x payloads. Is it only ninjhax?

Hikari-chin, 2017-06-19:
@maorninja No, they will not, cannot, and in my opinion, should not as they are completely unnecessary.

mitchellmebane, 2017-06-19:
The Rosalina menu shows the Luma3DS version - could it also show SysNAND/EmuNAND info (and the system version if there's room)?  That would obviate the need to patch System Settings.  Granted, EmuNAND is not particularly useful any more, but it still might be fun to mess with old system versions.

Also, would it be possible to add support for ZL/ZR in the menu key combo, or is that a technical limitation of how the New buttons are implemented?

bigretromike, 2017-06-19:
Ability to stream screen/video to external device.

figgyc, 2017-06-19:
Persist the app that loads HBL across reboot if you change it for convenience (might be useful say if at some point Nintendo does banwave by titles that may include hblauncher_loader and could be avoided by masquerading HBL as another useless app like AR Games) so you don't have to set it on each boot (unless this is meant to happen and I'm just messing up?)

Gudf, 2017-06-19:
@mitchellmebane I think it's a technical limitation. I can't find the issue where it's been mentionned, but it has already been suggested.

Hikari-chin, 2017-06-19:
Have gotten a couple feature requests that have either already been asked for or are already implemented. Everyone, please actually read the body of the main post before commenting.

maorninja, 2017-06-19:
How about, being able to swap out your ROMFS anytime?

Basically, I have a lot of NSMB2 hacks, and I like to play them all. But to do so, I have to go into FBI, rename the folders, then go and play. Why not just make a "ROMFS Selector" thingy?

HoLLy-HaCKeR, 2017-06-19:
@maorninja there wouldn't be much demand for that, it'd be easier to just code up a little application to switch it.

pechurc, 2017-06-20:
I've tried do "Brightness adjustment" on rosalina but I can't get GSPLCD functions working
Rosalina just turn to black screen and the system freezes, no .dmp file generated
You can try with this, GSPLCD_SetBrightness also is not working

```
if (R_SUCCEEDED(gspLcdInit())) {
		GSPLCD_PowerOffBacklight(GSPLCD_SCREEN_BOTH);
		gspLcdExit();
}
```
Sorry if this is no the correct place for comment this.

doesthisusername, 2017-06-20:
TAS tools, like save states (not sure if feasible at all), frame advance, and queued inputs. Probably not fit for the "main version" of Luma, though. So if it isn't, a plugin kind of thing to make it possible to implement would be nice.

Macros could also be fun.

DMSalesman, 2017-06-20:
Would it be possible for Rosalina to make the 3DS stay connected to an access point that cannot communicate to the Internet?

I'm asking this because e.g. FBI can receive .cia files and the like from a device connected to the same network as the 3DS, but at the moment an access point with Internet access is still required due to the above, and such an AP may not be always available.

WubTheGame, 2017-06-21:
Would it be possible to inject Any title into the currently running title? i.e Injecting Homebrew CIA's into 3DS Sound

onaryc, 2017-06-22:
Could it be possible to have a configuration file or different mode/profile choose entry (dev mode with debug/process/... related entries and a "normal" mode with non dev features) to configure the entries in the rosalina menu?

Anyway i think rosalina is a great feature! Great work!

Hyperom, 2017-06-23:
As I do a lot of Smash 3ds modding, I think the ability to un/remount the SD Card would be Incredibly useful. Although even i can see that this is a monumental request, but just saying, it would be nice to have.

This and possible a note taking ability. I would love to have this so as to write notes of what to do either in a game or when modding Smash 4.

HoLLy-HaCKeR, 2017-06-23:
@Hyperom I've heard that pen and paper work really well for notes.

astronautlevel2, 2017-06-23:
Posting this here because it's mostly relevant;

@LiquidFenrir and I have been working on adding in support for detecting new Luma3DS releases in Rosalina and sending a system notification, progress can be seen at https://github.com/astronautlevel2/Luma3DS-1

RubberBrony, 2017-06-24:
Would it be possible to allow the user to change certain aspects of the Luma3DS Configuration from within Rosalina?

Specifically, enabling or disabling game patching, showing NAND or user string in system settings, the GBA boot screen in patched AGB_FIRM, and other options could go in the menu. It could replace the New 3DS Menu (which currently only changes the clock speed and access to L2 cache, both options found in the Luma3DS Config menu on boot). 

Since it would be an options panel, some already planned and considered features such as screen brightness selection, volume control, and Wi-Fi status could potentially be in there as well.

Rosalina is a very impressive achievement, by the way! I love the expanded functionality of the HBL (uninstalled all the homebrew titles from my 3DS!); taking screenshots, messing with the clock speeds, and Input Redirection are absolutely incredible.

Hyperom, 2017-06-24:
@HoLLy-HaCKeR They're looking for requests, thats a request. Just doing my civic duty.

Hikari-chin, 2017-06-24:
@Hyperom This is actually already possible, with a homebrew called SDmounty.

LiquidFenrir, 2017-06-24:
@Hyperom to add to @Hikari-chin's answer, you don't even need SDmounty, just press L+R on the HOME menu to get in the camera applet and you can safely remove the SD card there. Don't answer to their prompt until the SD is back in though.

killer983, 2017-06-26:
Support for image capture similar to ntr cfw or horizonM

mariogamer2, 2017-06-26:
It would be great to support screen capture yeah.... like HorizonM since he has compat. with old 3ds but with better compatibility...

WubTheGame, 2017-06-26:
@killer983 and @mariogamer2 those are already in Rosalina, using the Take Screenshot feature (it takes several seconds to take the shot though). if you were talking about recording video however, i don't know if that's likely to happen or even be possible.

WubTheGame, 2017-06-26:
i just referenced that so that the op of that post can see this post.

Shadow322, 2017-06-27:
Have it to where screenshots r saved to sd card instead of ctrnand when u have boot.firm only on ctrnand

MarioMasta64, 2017-06-27:
a delete option for installed titles would be pretty neat. sometimes i go into fbi and look through so many titles then i forget what im looking for. i suppose going home and returning will solve it but itd be pretty neat to delete from home menu.

WubTheGame, 2017-06-28:
Any possibility that using Rosalina, you can use HBL kinda like an Applet? i.e Open Rosalina menu, go to Launch HBL (or whatever it would be named) and you could run a 3DSX without having to close your current software? this would be really useful for something like this: Playing a game that you want to spend play coins on, opening Rosalina menu after running out of play coins, and run HBL to use a Playcoin Generator and exiting out without closing the Game and now being able to spend more play coins. i Completely understand if this can't or won't be implemented. (also, i'd love to get home menu notifications about new stable luma releases)

DeckOfNoobs, 2017-06-28:
The ability to load custom ROMFS/EXEFS into NAND applications, like SD games, would be pretty neat.

TheMCNerd2017, 2017-06-29:
I personally would like to see more details about processes in the Process List, like the current CPU and RAM usage of a process.

Bolt23, 2017-06-29:
Virtual RAM for the 3DS to improve in-game performance? 1GB will be great rather than 63.9 MB? No? Na! Yeah JPN version is annoying! ;P

[65 more replies]

closed 2017-07-30 by Hikari-chin