LumaTeam/Luma3DS #621: Inconsistent/erroneous C-nub movement when using InputRedirection

Terathiel, 2017-06-16:
**System model:** New 3DS

**SysNAND version (+emu/redNAND version if applicable):** 11.2.0-E SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):** B9S 1.2

**Luma3DS version:** https://github.com/AuroraWright/Luma3DS/commit/e6e9f95c

**Luma3DS configuration/options:**

Default EmuNAND: ( 1 )

Screen brightness: ( 1 )

Splash: ( Off )

PIN lock: ( Off )

New 3DS CPU: ( Off )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Use custom path: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: (x)

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

--

**Explanation of the issue:**

Not sure if this is a Luma issue or a problem with Tux's InputRedir client, but I get unpredictable results with the second stick ("c-nub") when using my Logitech F310. Particularly noticeable when using diagonals, the console will sometimes register input in the opposite direction intended, often after I've switched quickly from one diagonal to its opposite. It doesn't seem to be an inverted axis problem since it's not consistent - axes register in the correct direction much of the time. I've experienced this both in the Home Menu and Final Fantasy Explorers.

**Steps to reproduce:**

1. Activate InputRedirection

2. Move the second analog (ie "c-nub"), especially from diagonal to diagonal

Jan200101, 2017-06-17:
could also be a controller Issue

TheNain38, 2017-06-18:
I also got this issue with my XBox360 controller

TheNain38, 2017-06-18:
I pushed a fix to Tux's InputRedirection Client: TuxSH/InputRedirectionClient-Qt#15

Terathiel, 2017-06-19:
Thanks for your investigation and prompt contribution! I suppose I should close this now, as it's technically resolved. Just have to wait for Tux to post a new build...

closed 2017-06-19 by Terathiel