LumaTeam/Luma3DS #622: region emulation doesn't work with some SNES VC titles

Jimmy-Z, 2017-06-16:
**System model:**

new 3DS LL (JP)

**SysNAND version (+emu/redNAND version if applicable):**

11.4.0-37J SysNAND, 11.4.0-37U emuNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap 1.2

**Luma3DS version:**

8.0-174089e4 TuxSH's build from #541

**Luma3DS configuration/options:**

Default EmuNAND: ( 1 )

Screen brightness: ( 1 )

Splash: ( off )

PIN lock: ( off )

New 3DS CPU: ( off )

Autoboot EmuNAND: ( no )

Use EmuNAND FIRM if booting with R: ( no)

Enable loading external FIRMs and modules: ( no )

Use custom path: ( no )

Enable game patching: ( yes )

Show NAND or user string in System Settings: ( yes )

Show GBA boot screen in patched AGB_FIRM: ( no )

Patch ARM9 access: ( no )

Set developer UNITINFO: ( no )

**Explanation of the issue:**

some out of region SNES VC titles (3 out of 4 I've tested) don't launch, raise an exception upon launch:
````
Processor: ARM11 (core 1)
Exception type: prefetch abort (svcBreak)
Current process: loader (0000000000000000)
````

000400000f701000 KTR-N-UABE The Legend of Zelda™: A Link to the Past
000400000f701200 KTR-N-UACE Super Metroid™
000400000f702c00 KTR-N-UAQE Contra III The Alien Wars

they fail exactly the same, with or without region emulation, crash dumps attached below.

they work in region changed emuNAND.

this one works without region emulation:
000400000f700e00 KTR-N-UAAE Super Mario World™

corresponding SFC VC titles work:
000400000f700f00 KTR-N-UABJ ゼルダの伝説 神々のトライフォース
000400000f701100 KTR-N-UACJ スーパーメトロイド
000400000f702b00 KTR-N-UAQJ 魂斗羅スピリッツ
000400000f700d00 KTR-N-UAAJ スーパーマリオワールド

and my region emulation setup(`USA EN US 01`) works with Majora's Mask US cartridge.
0004000000125500 CTR-N-AJRE The Legend of Zelda™: Majora’s Mask 3D

**Steps to reproduce:**

1. launch said title.

2. the 3DS loading animation shows up, but then the exception raises.

**Dump file:** including 3 dumps, in the order I listed them:
[crash_dumps.zip](https://github.com/AuroraWright/Luma3DS/files/1079944/crash_dumps.zip)

Jimmy-Z, 2017-06-16:
IIRC this used to work, I can't remember what titles I've tested back then but I used to have issues with some SNES VC and an injected VC, and region emulation can fix that:
https://gbatemp.net/threads/release-ultimate-snes-vc-injector-for-3ds.453891/page-7#post-7087947

And I believe region emulation is only required in the first run, after the initial setup, region emulation is not required, I still have that injected VC on my n3DSLL and it works without region emulation.

Hikari-chin, 2017-06-16:
Could you try this commit and tell me if the error still occurs? 

[boot.firm](https://u.nya.is/yynmqf.firm) - Just replace the boot.firm on your SD card root with this one.

Jimmy-Z, 2017-06-17:
no, still doesn't work.
[crash_dumps.zip](https://github.com/AuroraWright/Luma3DS/files/1082466/crash_dumps.zip)

ValiantBlade, 2017-06-19:
One game I have didn't work before the changes at all despite being multiregion. I have similar settings to Jimmy-Z, I have splash, pin lock, and n3DS CPU enabled (quite honestly exclusively for MGS Snake Eater 3D) but aside from that it's the same, and every time I opened Super Ghouls and Ghosts I got a crash before updating to 8.0, and now I get an exception instead. The game is the same for each region.

Jimmy-Z, 2017-06-19:
>  The game is the same for each region, it's all under the same titlekey.
It seems most USA SNES VCs are marked region ALL by that site. but this one is actually marked region USA.
000400000f702a00 KTR-N-UAPE Super Ghouls'n Ghosts™
SFC VC titles are different, it uses their Japanese names on that site, that might be the reason you missed it.
000400000f702900 KTR-N-UAPJ 超魔界村
I havn't tried it yet but I assume it will work, you should give it a try.

ValiantBlade, 2017-06-19:
I am USA region anyways so that just makes it even more confusing. Only EUR game I have is a DS game that worked before CFW and still works because the DS isn't region locked outside of China.

MYuitsu, 2017-06-19:
I have same issues with 	Return to PopoloCrois: A Story of Seasons Fairytale 000400000018CC00 USA, i have a New 3ds LL, 11.4.0-37U emuNAND, Boot9Strap 1.2, Luma3ds 8.0 release version. I changed many type of setting, or region, language in locale.txt, but i didn't work.

Jimmy-Z, 2017-06-20:
ValiantBlade you're running a USA SNES VC on USA 3DS so this is not related to region emulation.

MYuitsu your case is not SNES VC, and USA title on U region transfered emuNAND, this is not related to region emulation.

You two should open separate issues I'd say.

MYuitsu, 2017-06-21:
Jimmy-Z Thanks, I will post detail of my issue. And sorry for my mistake, my emuNAND is JPN,

ValiantBlade, 2017-06-21:
I am getting the same errors so the underlying cause for my issue might be the same. I'll keep watch on this thread and see if the resolution of this issue fixes it on my end. I have a decently large hunch it might be the same root cause somehow. Also the game is marked as all regions on Freeshop, using a recent titlekey file, so maybe the region on the site is mistaken, or maybe Freeshop?

Jimmy-Z, 2017-06-21:
@ValiantBlade so I tested Super Ghouls'n Ghosts on region transfered emuNAND, it works.

ValiantBlade, 2017-06-21:
Honestly at this point I'm just gonna use Snes9X for that game. Seems like a lot less of a hassle. I'm out of ideas on what's happening.

Jimmy-Z, 2017-06-21:
@ValiantBlade I used a CIA generated by FunkeyCIA.

Jimmy-Z, 2017-06-28:
Issue persist in 8.1

Hikari-chin, 2017-06-28:
So, I have been looking around, and all of the titles mentioned should be region-free, not marked as EUR, US, or etc. It's possible you've got some bad CIAs, but we won't provide support for software piracy here. 

Additionally, I tested all of the mentioned titles and did not have any issues launching them, locale emulation was not necessary.

closed 2017-06-28 by Hikari-chin

Jimmy-Z, 2017-06-29:
Those CIAs work in region transferred emuNAND, doesn't that prove they're not "bad"?

Jimmy-Z, 2017-06-29:
I suppose you tested in a EUR model and concluded "locale emulation was not necessary"

Hikari-chin, 2017-06-29:
Negative. My 3DS is a US region version.

Jimmy-Z, 2017-06-29:
So of course it won't need locale emulation in your situation, if you can't test it in JP model, you can try SFC VCs instead.

Hikari-chin, 2017-06-29:
Well, no, locale emulation wasn't necessary because these titles are region-free in the first place.

Jimmy-Z, 2017-06-29:
Region emulation is not all about region free.

I just tested two SFC VCs(Zelda and Metroid) in US emuNAND, they both raise a different exception on the first run, not the usual exception handler, no crash dump, had to long press power to power off, then they both run fine on the second run.

I suppose indeed you can't reproduce this if you don't have the willing to.

Jimmy-Z, 2017-06-29:
Region emulation and region free are different things. I don't know how to explain this exactly, but take a look at Wii U, loadiine/CFW/HAXCHI can run games from different region, but some titles needs some language files renamed to work, I suppose that's what region/language emulation about.

AFAIK Luma region free is always on, but some games needs region emulation.

Hikari-chin, 2017-06-29:
First off, my apologies for the initial misunderstanding. I had thought you were testing EUR titles, not US. 

I'm quite aware of what region free and region emulation are; I'm saying that these VC titles state support for US/EUR/JPN/KOR, thus they should work on those region systems. There shouldn't be a need to emulate any region with these titles. I will create a Japanese emuNAND and test this myself in a few moments. 

I attempted to launch the SFC titles you mentioned and got the exception screen that you did. However, upon putting a locale.txt with "JPN JP" for those titles, they booted up and launched without issue.

Jimmy-Z, 2017-06-29:
Thank you!

I have a friend with a new3DS(JP) still on A9LH Luma 6.6, I'll try to test if it was working like I remembered.

Jimmy-Z, 2017-06-29:
As for those SFC titles on US system, they work on second run, no region emulation needed.

Interesting these situations don't work vice-versa.

Jimmy-Z, 2017-06-29:
As for those SNES titles been region free, I think the last letter is a clue, "E" for US, "P" for EU, "J" for JP, those SNES VC titles are (on some way) intended for US I guess, "real" region free titles like Pokemon has "A".

Hikari-chin, 2017-06-29:
Okay. I had someone test this, and they confirmed that it worked on their JPN console running Luma 8.1. Here is what I had them do:

1. Installed the titles.
2. Create a /luma/titles/TID/locale.txt for them consisting of "USA EN" 
3. Enable "Enable game patching" in Luma's config.
4. Launch the titles

The tester was able to get them working without issue on their JPN console.

Jimmy-Z, 2017-06-29:
This is awkward, I reinstalled those CIA(same files I used days before), and they work now.

Even without title region emulation, crash on the first run, subsequent launches are fine, just like SFC VC titles on US system.

Downgrade to Luma 8.0, they still work.

Since the same behavior happens on 4 different titles, I never though it would be a crooked install.

Sorry for your wasted time :(