LumaTeam/Luma3DS #623: [Feature Request] Screen capture

bigretromike, 2017-06-16:
Is would be possible to include screen-capture like ntr have. This would be a great thing to enable with only one cfw used. Also maybe there would be a opportunity to improve it.

As there is a screen shot feature luma have access to screen 👍

PolarisBeaver, 2017-06-16:
There is one already in Luma 8.0

Access the menu by pressing L+Down+Select together. You will see the screenshot option there.

Jimmy-Z, 2017-06-16:
I guess he meant screen streaming like NTR, he mentioned screenshot only to prove it should be possible.

That's the most f*cked up English I've ever read.

PolarisBeaver, 2017-06-16:
Oops I misread it seems. But yes, capture would be great. I'm guessing O3DS can't have this option anytime?

bigretromike, 2017-06-16:
@Jimmy-Z yeah, a lot of typos; I fixed it (i think). and yes, I'm talking about video capture like the one in NTR.

@PolarisBeaver who knows. Maybe Luma would be able to achieve partial support (with less cpu demanding games)

Hikari-chin, 2017-06-16:
Closed. Feature requests for Rosalina go in #619 only.

closed 2017-06-16 by Hikari-chin