LumaTeam/Luma3DS #624: Screen Errors, and Crash from Rosalina

Kaisogen, 2017-06-17:
**System model:**

New 3DSXL

**SysNAND version (+emu/redNAND version if applicable):**

11.4 SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap 1.2

**Luma3DS version:**

Stable 8.0

**Luma3DS configuration/options:**

Default EmuNAND: ( 1)

Screen brightness: ( 4)

Splash: (Off)

PIN lock: (4 )

New 3DS CPU: (Clock + L2 )

--

Autoboot EmuNAND: ( No)

Use EmuNAND FIRM if booting with R: ( No)

Enable loading external FIRMs and modules: (No )

Use custom path: ( No)

Enable game patching: (Yes )

Show NAND or user string in System Settings: ( Yes)

Show GBA boot screen in patched AGB_FIRM: ( Yes)

Patch ARM9 access: ( No)

Set developer UNITINFO: (No )

--

**Explanation of the issue:**

When playing one of the DLC maps (Wealth before Health) on Fire Emblem Echoes, you can mess up the screen. I don't know how to describe it, but the screen is filled with white lines on both of them. The game is fully functional, the screen just isn't showing properly. You can also crash the game by entering and exiting the Luma Menu over and over.  

**Steps to reproduce:**

1. Open the DLC map (Wealth Before Health) in Fire Emblem Echoes. I haven't tested if this works with multiple maps, but it is reproduceable.

2. Open the Rosalina Menu, and go into New 3DS Settings.

3. Close the menu and return back to the game. There will be a chance the screen error will happen. It seems to be entirely random.

Crashing the system..

1. Open the game (this can be done from the map)

2. Spam open and close the Rosalina Menu.

3. It will crash randomly

**Dump file:**
I'll upload the one for the second issue later.

TuxSH, 2017-06-17:
[boot.zip](https://github.com/AuroraWright/Luma3DS/files/1082208/boot.zip)
What about the latest commit here?

RS55, 2017-06-17:
TuxSH This problem and the screenshot issue has been fixed, apparently in the last commit. Yay!. I've spamed the combo button and taken screenshot, no problem at the moment. Before I was afraid to take a screenshot or open rosalina hahaha. Sorry if I have some grammatical error, I speak Spanish lol

closed 2017-06-17 by TuxSH