LumaTeam/Luma3DS #625: Luma Hourly arent work

GameStepHD, 2017-06-17:
**System model:**

[e.g. 2DS, New 3DS, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**

[e.g. 11.4.0-37U SysNAND, 11.4.0-37J EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**

[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**

[e.g. 7.1 stable or if using nightly/hourly specify the commit like this https://github.com/AuroraWright/Luma3DS/commit/9570e6cbeca53128433abbf5e3473cb8a07fe69e]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: (4 )

Splash: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Use custom path: ( )

Enable game patching: (x )

Show NAND or user string in System Settings: ( x)

Show GBA boot screen in patched AGB_FIRM: ( x)

Patch ARM9 access: (x )

Set developer UNITINFO: ( )

--

**Explanation of the issue:**

The Luma3DS Nightlys arent working on my 3ds

**Steps to reproduce:**

1. update to hourly

2. 3ds boots to blackscreen with Power LED on

**Dump file:**

ahoischen, 2017-06-17:
Please fill in the system model, version and entry point fields. Are you on b9s 1.2?

GameStepHD, 2017-06-17:
new 3ds xl on 11.4.0-37E
yes its bs1.2

SvenDaHacker64, 2017-06-17:
Hourly build website is broken. You have to manually compile their project yourself. https://github.com/AuroraWright/Luma3DS#compiling
Requires devKitPro and Git Bash

I could give you one.
[Luma3DS v8.0-2363817.zip](https://github.com/AuroraWright/Luma3DS/files/1082591/Luma3DS.v8.0-2363817.zip)

ahoischen, 2017-06-17:
Can't confirm with 2363817 using a n3ds with splash after boot, user string, and patched GBA firm. @GameStepHD could you please clarify which commit you are talking about?

GameStepHD, 2017-06-17:
https://github.com/AuroraWright/Luma3DS/commit/901d4992c0d702ae451fd12785563da448ed3d83

ahoischen, 2017-06-17:
That hourly is already old and has been reverted immediately afterward. I think this issue can be closed since it's fixed in all later builds.

GameStepHD, 2017-06-17:
i used these https://astronautlevel2.github.io/Luma3DS/

ahoischen, 2017-06-17:
901d4992c0d702ae451fd12785563da448ed3d83 has been reverted by 69762c5dce82534458ff66d45764a91d859e6341 8 days ago. The website you are using is the hourly website and as @SvenDaHacker64 said:
> Hourly build website is broken.

I could now dig around and search for the issue which caused 901d4992c0d702ae451fd12785563da448ed3d83 to be reverted and link that here but I am pretty certain this is a duplicate without doing so. Even if it is not, the problem has been fixed already so there is no point keeping this issue open.

closed 2017-06-17 by GameStepHD