LumaTeam/Luma3DS #626: Fire Emblem Fates Crash after Battle report

D4rkelton, 2017-06-17:
**System model:**

Old 3DS

[e.g. 2DS, New 3DS, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**

11.4.0-37E

[e.g. 11.4.0-37U SysNAND, 11.4.0-37J EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):*Boot9Strap*

[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**
8.0 Stable
[e.g. 7.1 stable or if using nightly/hourly specify the commit like this https://github.com/AuroraWright/Luma3DS/commit/9570e6cbeca53128433abbf5e3473cb8a07fe69e]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: (4)

Splash: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Use custom path: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

--

**Explanation of the issue:**
Fire Emblem Fates Revelations crashes after the first section off the credits

**Steps to reproduce:**

1. Play through the final Level of Fire Emblem Fates Revelations (not sure if ocurring with other versions)

2. Watch the first part of the credits. (Battle Summary)

3. Crash

**Dump file:**

[Crash.zip](https://github.com/AuroraWright/Luma3DS/files/1082617/Crash.zip)

hedgeberg, 2017-06-19:
Can you reproduce the error on a different level or on a clean save file? If so, I can try to reproduce the error. Allternatively, can you dump you save and tell me how you dumped it + post it here? I can load it up and try to see if i can make it happen as well. As is, unless we have some datapoints beyond just your unit, we don't have a good way of pinpointing the issue. thx!

Hikari-chin, 2017-06-26:
Going to close this; I cannot reproduce this, and requests for further information have been left unanswered for just about a week. Will reopen if this is shown to be an actual issue at a later point.

closed 2017-06-26 by Hikari-chin