LumaTeam/Luma3DS #627: 13 games Totally broken on start

chrisssj2, 2017-06-17:
New3ds EU
256 GB SD
11.4 sysnand and emunand combo unlinked
Bootstrap 1.2
latest official luma with game patching enabled. rest default options

-Puzzle and dragons mario bros edition Exception type data abort
-Heroes of Ruin (errdisp) error typed logged process id 39 processname riftwalk, adress 0x003ec3dc error code : 0x8a0f832
-IronFall (prefetch abort svc break
-Tank troopers (no other info at crash)
-Bust a move (black screen but able to go to home menu so not exactly crash)
-Disney Magical World (black screen but able to go to home menu so not exactly crash)
-Tekken 3D ( data abort)
-Team Kirby Clash Deluxe (black screen but NOT able to to home menu)
-Zelda major's mask (data abort)
-Pokemon Rumble blast ARM11 prefetch abort svcbreak
-Dragon Quest VIII An error ocured while the sd card was being written to please turn off the system and re-insert the sd card. please close the software or turn the system off
-Mario vs donkey kong tippings stars (data abort)
-Paper mario sticker star Errortype (errDisp) Error type Generic Adress 0x002a1064 Error code 0xc8804470

I can reproduce this easily each time i start these games, re-installing them have no effect.

HugaShire, 2017-06-17:
Try turning off game patching. Also try this build
[https://github.com/HugaShire/Luma3DS-Nightlies/raw/master/boot.firm](url)

chrisssj2, 2017-06-17:
First i try your suggested build: no change
Then i turn off game patching with the new build: no change either. still crashes.

HugaShire, 2017-06-17:
Try a new or different SD?

chrisssj2, 2017-06-17:
This SD is just newly bought. But for testing argument i can use a smaller spare sd and try the game.

Hikari-chin, 2017-06-17:
Try turning off the New 3DS clock+L2 boost either in the config menu or in Rosalina's New 3DS menu. 

Can't personally reproduce this, though.

chrisssj2, 2017-06-17:
Just tested with a new sd still same result. The New 3DS clock+L2 boost was already off.
How are you not able to reproduce this? have you tried any of the games i listed?

TWonderchild, 2017-06-17:
"-Pokemon Rumble blast"
"-Zelda Majora's Mask"
I only own these two but they are working on my 2DS with Luma 8.0-901d4992. No game patching enabled. Are those non funtional games on your DS maybe not for your DS's region? Region patching is not working properly according to some ppl on reddit. Maybe try reinstalling those games.

Derek1199, 2017-06-18:
@chrisssj2  do you have an emunand? Or ever region changed? Or made some mistake following the 3ds guide ever?

Kalanyr, 2017-06-18:
I can verify that DQVIII (US) is working fine for me on a US N3DS on 11.4 with Luma 8

Tekken 3D (EU) is working fine on an equivalent EU N3DS. 

Don't have any of the other games listed installed. 

You may want to verify that your installs aren't corrupted (either pre or post installation).

Kalanyr, 2017-06-18:
I should note that's with CPU and LVL2 clock enabled for both.

closed 2017-06-18 by Hikari-chin

Derek1199, 2017-06-18:
Maybe your Ticket.db or title.db has an issue @chrisssj2

chrisssj2, 2017-06-18:
I have an emunand yes. What do you mean with region change? And no i did not make a mistake with 3ds guide. I wiped my sysnand with tinyformat a week ago, and restored my system with an old nand backup.

And I verified that my backups arent corrupt since i installed via freeshop for example, uninstalled app and re installed with eshop. 

Any other suggestions?

chrisssj2, 2017-06-18:
I think i solve some of it.. Seems re-installing the game in EUR (all games that fail are USA)
Will make them atleast boot.
However 1 game crash @main menu when you say no to all the streetpass questions etc.
Heroes of Ruin (errDisp)
errortype logged 
adress 0x003ec3dc
0xc8af833

And then I still have this problem: Which I tried using luma localeswitcher for, but it still keeps getting that message..

Dragon Quest VIII
An error ocured while the sd card was being written to please turn off the system and re-insert the sd card.
please close the software or turn the system off

And then some other issue I noticed:

-Shin megami Tensei IV
The addon content could not be recognised
-Shin megami Tensei IV Apolcalypse
The addon content could not be recognised

-Yoshi island unable to create new data on sd card.

FaeMinx, 2018-07-27:
I found this because I was getting that error for DQVIII.

European N3DS,
Cart was bought in America (because).
128Gb MicroSD, CFW (you know the story)

The solution was to use FBI to determine the title of your offending game (starts 0004, my DQVIII was 000400000018F100), then pop out the SD card, open it on a computer - navigate to your Luma folder, inside create a folder named 'titles'. Inside the 'titles' folder, create another folder named the title of your offending game (starts 0004). Inside that folder, create a text file 'locale.txt', and in the text file type (in caps) the region and language of your game.... 
example: 'USA EN' (which is what I put for my American DQVIII).

So in my case, the path looked like this:
"\luma\titles\000400000018F100\locale.txt" 

Next step, put the SD card back in your 3DS, hold 'Select' when powering on your 3DS to get into Luma menu, and then make sure you enable game patching.

Press 'Start' to save and exit, and then try out your game!

mavaje, 2019-08-06:
I had an issue with a USA N3DS with b9s running retail EUR Majora's Mask consistently crashing during the initial loading before the menu. After trying almost everything else I was able to fix it by following FaeMinx's solution of adding /luma/titles/0004.../locale.txt containing "EUR EN".
Thanks for the perfect solution!

limaer, 2020-02-17:
> I found this because I was getting that error for DQVIII.
>
> European N3DS,
> Cart was bought in America (because).
> 128Gb MicroSD, CFW (you know the story)
>
> The solution was to use FBI to determine the [...]

OMG will you marry me?? I've been agonizing over this forever and your solution worked perfectly :D

InfectedRat, 2022-09-01:
> I found this because I was getting that error for DQVIII.
>
> European N3DS, Cart was bought in America (because). 128Gb MicroSD, CFW (you know the story)
>
> The solution was to use FBI to determine the [...]

> I found this because I was getting that error for DQVIII.
>
> European N3DS, Cart was bought in America (because). 128Gb MicroSD, CFW (you know the story)
>
> The solution was to use FBI to determine the [...]

It really works! I was looking for a lot of information about the launch of this game until I came across this advice. Thank you so much.

Yorper, 2023-09-16:
> I found this because I was getting that error for DQVIII.
>
> European N3DS, Cart was bought in America (because). 128Gb MicroSD, CFW (you know the story)
>
> The solution was to use FBI to determine the [...]

You are a legend! This worked for me with an Australian Paper Mario on a Japanese system.

Sky2310, 2024-01-21:
The following is how to fix Arm11 exception has occured errors causing some games installed from CIA file to not boot:

1) Ensure the latest Luma 3DS boot files have been loaoded - https://github.com/LumaTeam/Luma3DS/releases by copying/replacing boot.3dsx and boot.firm files in root of SD Card

2) After installing the CIA game file, go back to FBI App > Titles > navigate to the affected game (e.g. TANK TROOPERS), select Import Seed then Home to Exit (with internet connection on). If no internet connection or auto online seed-loading failed (manual seed download) go to: https://hshop.erista.me/c/games > search for the game title (make sure Game ID matches) > Download the Seed for FBI (then copy the .dat file (should look something like 000400000017CC00.dat) to sd:/fbi/seed folder, repeat the above Import Seed step (with internet connection turned off).

Credit: DukeMingus https://www.reddit.com/r/3dspiracy/comments/dn1fuh/error_code_on_certain_games/

3) Put SD Card into PC, go to Luma folder and create a new folder called 'titles' (no quotation marks), then create another folder named exactly like the Game ID (Game ID displayed while installing with FBI or listed on https://hshop.erista.me/c/games, now in that folder create a blank .txt file (with Notepad) named 'locale.txt', go inside it and type: 'USA EN' (not quotation marks) - USA relates to region of game and EN refers to English title release (for USA region games), EU (for European region games which also includes Australia and New Zealand), JAP (for Japanese region games) and KOR (for Korean region games), finally put SD Card back into 3DS an, hold Select and then power the console on > Luma3DS Menu should appear - makesure 'Enable Game Patching' is selected then Press Start to save and exit.

Credit: FaeMinx https://github.com/LumaTeam/Luma3DS/issues/627/

The affect game should now boot up perfectly - only affect consoles on the latest firmware 11.17.0-50 / May 22, 2023.

_Sky_

PabloMK7, 2024-01-21:
Deleted @Sky2310 comment as it contained piracy assistance.