LumaTeam/Luma3DS #630: "An Exception Ocurred" when trying to load a Terraria 3DS world.

Franvcg, 2017-06-18:
[Old 3DS XL]

[11.4.0-37U SysNAND]

[Boot9Strap]

[Luma v8.0]

**Luma3DS configuration/options:**

Default EmuNAND: (1)

Screen brightness: (4)

Splash: (Off)

PIN lock: (Off)

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Use custom path: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (X)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

--

**Explanation of the issue:**
I get "An Exception Ocurred" (Data abort) every time I try to load my world on Terraria 3DS, the first time it happened was after I closed terraria without saving (Home button -> Close software). I've created a new world and I can load it without errors.

**Steps to reproduce:**

1. Close a terraria world without saving (Home button -> Close software)

2. Try to load it again

**Dump file:**
[crash_dump_00000000.zip](https://github.com/AuroraWright/Luma3DS/files/1083318/crash_dump_00000000.zip)

pbanj, 2017-06-18:
tbh im not sure this is a luma issue. last time i tried to play terraria it screwed my entire system up until i uninstalled it.

Hikari-chin, 2017-06-19:
I can't reproduce this on my end, but being unable to load one world but loading others fine points to save data corruption; this is very unlikely to be a Luma issue.

closed 2017-06-19 by Hikari-chin