LumaTeam/Luma3DS #631: Current Commit changes Version to 0.0.0-0 in config menu

HugaShire, 2017-06-18:
**System model:**
2DS
[e.g. 2DS, New 3DS, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**
11.1.0-33U Sys
[e.g. 11.4.0-37U SysNAND, 11.4.0-37J EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**
Boot9Strap 
[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**
https://github.com/AuroraWright/Luma3DS/commit/17828273a5a9450658fc36555cf7892ded7e65b2
[e.g. 7.1 stable or if using nightly/hourly specify the commit like this https://github.com/AuroraWright/Luma3DS/commit/9570e6cbeca53128433abbf5e3473cb8a07fe69e]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Use custom path: ( )

Enable game patching: (x)

Show NAND or user string in System Settings: (x )

Show GBA boot screen in patched AGB_FIRM: (x)

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

--

**Explanation of the issue:**
In the commit I specified, the Luma3DS Config menu will say "Luma3DS v0.0.0-0 configuration.

**Steps to reproduce:**

1.Press select on boot

2.Read the info it gives.

**Dump file:**

pbanj, 2017-06-18:
mine reads just fine

HugaShire, 2017-06-19:
Hmm, I tested one from the nightly site, and it works, guess it was my compiling.

TuxSH, 2017-06-19:
That means you just don't have git correctly installed/set-up @HugaShire

HugaShire, 2017-06-25:
@TuxSH. It was. Thanks for that tip.

closed 2017-06-25 by HugaShire