LumaTeam/Luma3DS #632: "The card was removed" error

UraKn0x, 2017-06-19:
**System model:**

New 3DS 11.3.0-36E

**Entrypoint (How/what you're using to boot Luma3DS):**

[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**

8.0

**Luma3DS configuration/options:**

Default EmuNAND: (1)

Screen brightness: (4)

Splash: (Off)

PIN lock: (Off)

New 3DS CPU: (Any, but currently off)
--

Autoboot EmuNAND: (Off)

Use EmuNAND FIRM if booting with R: (Off)

Enable loading external FIRMs and modules: (Off)

Enable game patching: (Off)

Show NAND or user string in System Settings: (On)

Show GBA boot screen in patched AGB_FIRM: (On)

Patch ARM9 access: (Off)

Set developer UNITINFO: (Off)

Disable Exception Vectors: (Off)

--

**Explanation of the issue:**

I updated b9s to 1.2 and Luma to 8.0, everything worked fine until it suddenly started throwing "The card was removed" errors every time I try to launch a title that's not installed on the NAND. Even GBA, DS/DSi games and 3DS cartridges throw me the error.
It does work fine if I go back to a previous version of Luma though...

**Steps to reproduce:**

1. Unknown, it started doing that for no reason.

**Dump file:**
None, this error doesn't produce a dump file.

Derek1199, 2017-06-19:
I've never had this issue before but I can give you the latest commit if you want

Derek1199, 2017-06-19:
nvm I can't

UraKn0x, 2017-06-19:
Well, I can get astronautlevel12's latest nightly, but it has the same problem anyway…

Derek1199, 2017-06-19:
Oh. It's probably some other issue. Did you ever use a really outdated firmware?

Derek1199, 2017-06-19:
and do you have a o3ds? Try installing the TWL and AGB firms if you did.

UraKn0x, 2017-06-19:
I'm on N3DS, and I didn't install A9LH until OTPless was available, so the latest fw I've ever been on was the one it had when I bought it, so 9.0 or 9.2.

Derek1199, 2017-06-19:
did you ever region change?

UraKn0x, 2017-06-19:
Only on EmuNAND, and I don't even think it was on this 3DS anyway…

Derek1199, 2017-06-19:
were your NAND's linked? It is important that you are very descriptive on everything you have done in the past. So that you can receive proper help.

Derek1199, 2017-06-19:
I would suggest trying to install the TWL and AGB firm's. This may help you.

UraKn0x, 2017-06-19:
Nope, I've always had my NANDs unlinked because I knew from my previous 3DS that linked NANDs were a pain. I also used GW in the past before switching to A9LH, but I never used their A9LH implementation/downgrade thing. I don't think the problem is related to something I've done in the past though as it worked for several days on 8.0 before suddenly breaking... I haven't done anything weird between updating to 8.0 and the moment it broke, I only tried a couple of new features out, then played some games, at this point it still worked, and then one day I turned it on to play and none of the games would launch.

Derek1199, 2017-06-19:
I can't help you then. Best to ask tuxSH about this.

UraKn0x, 2017-06-19:
btw there's apparently someone on the Nintendo Homebrew Discord who has (had? idk) the same problem

Derek1199, 2017-06-19:
I wouldn't know. I got banned from that server months ago. Sorry.

UraKn0x, 2017-06-19:
Haha it's no problem, I just wanted to say it's apparently not a unique case

Trojaner1982, 2017-06-19:
had the same problem with "SD Removed" ... Fresh formated the SD and installed all apps/games fresh ... all work fine now ... still on Luma 8.0 ... some other ARM Crash Problems form apps/games gone too

UraKn0x, 2017-06-19:
Did it throw the error with game cartridges though?

Trojaner1982, 2017-06-19:
never tried a Card since i´ve dumped all my cardridges

Hikari-chin, 2017-06-19:
I can't reproduce this with any of my 3DS carts, DS carts, or DS flash carts. Could you try the latest commit [here](https://u.nya.is/mxgcqp.firm), please, and let me know if the issue persists?

UraKn0x, 2017-06-19:
Hmm, it fixed the issue... that's weird, but cool I guess, thanks

closed 2017-06-19 by Hikari-chin

gingerbeardman, 2019-02-01:
Is this fix in the latest Luma? @yuukieve

Dcortese02, 2024-09-20:
How do I replace the boot.firm file. Im having the same issue after using FTPd

AsureDawn, 2025-10-18:
I'm not sure what caused it, but I also encountered this after installing zelda: oot3d to nand. I exited godmode and moved it to my zelda folder, then I noticed majora's mask didn't have the graphic on the top screen when I went over it. Booting Majora's mask failed with "sd card removed". It may have happened before installing oot, but I hadn't noticed. It was certainly working when I had first installed it.

I couldn't find it under titles in FBI, and there was a single unnamed software in system management. I attempted to remove the unnamed software there, but it supposedly failed. However, MM was not in my zelda folder anymore.

So, after all that, I booted up godmode again and reinstalled MM to nand. The game is now installed successfully and runs fine. Unfortunately, I hadn't created a save, so I don't know if that would be preserved if I had to do this again. I'll post an update here if it comes to pass.

This is with a recently modded original 3ds, so it should be on the latest luma. System software is updated to latest now, was on an earlier version when I modded it though. Never flashed older fw or anything like that. Rosalina says:
Luma3DS v13.3.3
Kernel ver. 2.58-0
MCU FW ver. 2.37
PMIC Vend. 0
Batt Vend. 0