LumaTeam/Luma3DS #633: 3DS Stuck trying to shut off when i install a cia file

Seba201x, 2017-06-19:
**System model:**
NEW 3DS
[e.g. 2DS, New 3DS, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**
11.4.0-37 SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**
Boot9Strap

**Luma3DS version:**
8.0 Stable

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: (After)

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Use custom path: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (X )

Show GBA boot screen in patched AGB_FIRM: (X)

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

--

**Explanation of the issue:**
When i install a cia file and i want to shut down my 3ds, the 3ds doesn't shut down, i have to press the power button for several seconds to turn off the 3ds

**Steps to reproduce:**

1.Install any cia file with fbi

2.Turn off the 3ds

ZenIsBestWolf, 2017-06-19:
Cannot reproduce whatsoever. Possibly corrupt CIA?

Seba201x, 2017-06-19:
@ZenIsBestWolf i don't believe that, i tried even with some titledb cias.

closed 2017-06-19 by TuxSH