LumaTeam/Luma3DS #634: Card error. System restart, lead to black screen on boot

Sorablue89, 2017-06-19:
**System model:**
Old 3DS XL

**SysNAND version (+emu/redNAND version if applicable):**
Sys 11.4.0-37U

**Entrypoint (How/what you're using to boot Luma3DS):**
Boot9strap

**Luma3DS version:**
8.00

**Luma3DS configuration/options:**
I forgot and can't enter the config

**Explanation of the issue:**
I installed boot9strap and luma 8.00 since last week. No issues happened. But today somehow i get error "card removed. system will restart" or so and since that my 3DS can't boot nor enter luma config screen, it just stay on black screen when turned on. I add screenshot of my SD card content to see if there is something that i am missing or no.
[image]

**Steps to reproduce:**
Can't reproduced

**Dump file:**
No dump file

retrofank, 2017-06-19:
This is not a support page and by the looks of it, it's mostly likely a SD card failure, not Luma's fault.  

You should raise this in the relevent support forums.

retrofank, 2017-06-19:
Plus remove those a9lh bins.

TuxSH, 2017-06-19:
Very likely a SD card failure

closed 2017-06-19 by TuxSH