LumaTeam/Luma3DS #635: Mario kart 7 patching broken

Dark124, 2017-06-19:
**System model:**

[e.g. 2DS, New 3DS, Old 3DS]

**SysNAND version (+emu/redNAND version if applicable):**

[e.g. 11.4.0-37U SysNAND, 11.4.0-37J EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**

[e.g. Boot9Strap/Sighax, etc.]

**Luma3DS version:**

[e.g. 7.1 stable or if using nightly/hourly specify the commit like this https://github.com/AuroraWright/Luma3DS/commit/9570e6cbeca53128433abbf5e3473cb8a07fe69e]

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: (max)

Splash: (off)

PIN lock: (none )

New 3DS CPU: (off )

--

Autoboot EmuNAND: (off )

Use EmuNAND FIRM if booting with R: ( off)

Enable loading external FIRMs and modules: ( )

Use custom path: (off )

Enable game patching: (on )

Show NAND or user string in System Settings: (on )

Show GBA boot screen in patched AGB_FIRM: (off )

Patch ARM9 access: (off )

Set developer UNITINFO: (off)

--

**Explanation of the issue:**
Yes, I know what I am doing I've done it tons of times, it doesn't work anymore like it did before luma 7.X, I have everything in the right place and I check everything a bunch of times

**Steps to reproduce:**

1.

2.

**Dump file:**

TuxSH, 2017-06-19:
Not giving support for 7.x, and insufficient informatio,.

Also use latest commit (8.0 has a bug regarding langemu).

Jan200101, 2017-06-19:
There seems to be a lack of Reproduction steps.
I'm gonna go ahead and say this is  user error

Hikari-chin, 2017-06-19:
In addition to what @TuxSH said, have you changed anything about your modifications recently? Updates to MK7, updates to the patches, whatever? Or is updating Luma from 7.1 to 8.0 the only factor that's changed?

Dark124, 2017-06-19:
@Hikari-chin Luma update that is it

Dark124, 2017-06-19:
@TuxSH Yeah I know where it goes. :P

Dark124, 2017-06-19:
Okay for some reason this newer build works, even though I was on latest. I guess you built this from source

closed 2017-06-19 by Dark124

Hikari-chin, 2017-06-19:
@Dark124 Yes, it was built from source. The "hourlies" site used by Hamcha/Kunoichi's LumaUpdate is actually four commits out of date as of now.