LumaTeam/Luma3DS #636: Cannot watch 3D videos on Youtube

KiraYamatoSD, 2017-06-19:
**System model:**
New 3DS

**SysNAND version (+emu/redNAND version if applicable):**
11.4.0-37U SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**
Boot9Strap/Sighax

**Luma3DS version:**
8.0-21db0d4

**Luma3DS configuration/options:**
Default EmuNAND: ( )
Screen brightness: ( )
Splash: ( )
PIN lock: ( )
New 3DS CPU: (X)
--
Autoboot EmuNAND: ( )
Use EmuNAND FIRM if booting with R: ( )
Enable loading external FIRMs and modules: ( )
Use custom path: ( )
Enable game patching: ( )
Show NAND or user string in System Settings: (X)
Show GBA boot screen in patched AGB_FIRM: ( )
Patch ARM9 access: ( )
Set developer UNITINFO: ( )
--

**Explanation of the issue:**
Cannot watch 3D videos on Youtube.

[image]

**Steps to reproduce:**

1. Open Youtube
2. Search for "3ds movie 3d"
3. Scroll down till you find a video titled "3d 3d video 3ds 3d movie in 3d" by "3DVideoNature"
Any movie with left and right screen will do.
4. 3DS crashes

**Dump file:**
[dumps.zip](https://github.com/AuroraWright/Luma3DS/files/1085225/dumps.zip)

TuxSH, 2017-06-19:
Likely not a luma issue, also please disable all option before testing.

Stary2001, 2017-06-19:
I tried to reproduce (with that same video) and couldn't - it played with the 2 views side by side.

closed 2017-06-19 by Hikari-chin