LumaTeam/Luma3DS #637: Homebrew Launcher shutdown

gaetano2004pcc, 2017-06-19:
System model:

[Old 3DS]

SysNAND version (+emu/redNAND version if applicable):

[11.4.0-37U SysNAND]

Entrypoint (How/what you're using to boot Luma3DS):

[Boot9Strap]

Luma3DS version:

[8.0]

Luma3DS configuration/options:

Default EmuNAND: (1)

Screen brightness: (4)

Splash: (Before)

PIN lock: (Off)

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Use custom path: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (x)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

[image]

How do I fix it?

HugaShire, 2017-06-19:
What did you do to get it to crash?

Hikari-chin, 2017-06-19:
Are you using the [latest version of the Homebrew Menu](https://github.com/fincs/new-hbmenu/releases)?

gaetano2004pcc, 2017-06-19:
I'm using version 1.1.1 of hbl and every time I open a tool or any other
application for hbl give me this screen

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
Mail
priva di virus. www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

Hikari-chin, 2017-06-19:
Use the [latest version of the Homebrew Menu](https://github.com/fincs/new-hbmenu/releases). Closed.

closed 2017-06-19 by Hikari-chin