LumaTeam/Luma3DS #639: [Suggestion] Feature to change a console's DeviceID

Garro, 2017-06-20:
A feature to change the deviceID of a console would be extremely useful for those with the bypassing of the NNID service ban (known also as 022-2812).

This was researched back then with the Sun/Moon ban, there's a dump of different posts related to the ban here: https://gbatemp.net/threads/heres-what-we-know-about-002-0102-and-022-2812.449218/#post-6850004 (Including the trick related to messing up the RAM and the test related to the deviceID)

Monotek18, 2017-06-20:
It does not work anymore (the "ram" trick)
Anyway that a good way to be flagged.

Garro, 2017-06-20:
Any source on that?

Hikari-chin, 2017-06-20:
See #286. This particular method doesn't work anymore anyway, but I'll state this clearly: **Methods to bypass bans from Nintendo's Online Services will not be implemented in Luma3DS, period.**

closed 2017-06-20 by Hikari-chin