LumaTeam/Luma3DS #642: GBA titles crash most of the time/ error has occurred on load

cooldude413, 2017-06-21:
**System model:**

[ New 3DS]

**SysNAND version (+emu/redNAND version if applicable):**

[ 11.4.0-37U SysNAND, no EmuNAND]

**Entrypoint (How/what you're using to boot Luma3DS):**

[ Boot9Strap.]

**Luma3DS version:**

[8.0 stable] 

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Use custom path: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (yes )

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

--

**Explanation of the issue:**
Whenever I try to load gba games I sometimes get the an error has occurred issue and have to restart the system. I have tried all the troubleshooting steps both here and  3ds.guide and reinstalled both the agb and twl files but nothing worked. Moreover when I modified my girlfriends old 3ds everything worked perfectly. I have also tried to reinstall b9s and luma and still no luck. 

The games crash on first load sometimes, other times on 2nd, etc. It's pretty random. 

Any help would be greatly appreciated. 

**Steps to reproduce:**

1.load luma and launch gba game cia

2.

**Dump file:**

Margen67, 2017-06-21:
https://github.com/AuroraWright/Luma3DS/wiki/FAQ-and-troubleshooting#i-get-an-auto-shutdown-with-gbadsdsiware-games

cooldude413, 2017-06-21:
I have tried already to reinstall the agb firm and it didn't fix the issue. Interestingly enough, dsi ware games always work

Hikari-chin, 2017-06-21:
Could try a newer commit like [this one](https://astronautlevel2.github.io/Luma3DS/builds/Luma3DS-1782827.zip)? Just replace the boot.firm on your SD card with the boot.firm from this archive. 

However, I can't reproduce this issue even on 8.0, even hacked up GBA VC injections don't give me any issues.

DeathAngel74, 2017-06-21:
Pokemon Yellow , Emerald, and Emerald JPN work fine here.

DeathAngel74, 2017-06-21:
VC injects, btw.

cooldude413, 2017-06-21:
I'll try tonight. Also I'll upload the dump file. Maybe it might help, hacked gba roms crash much more often than official ones

DeathAngel74, 2017-06-21:
Try clean unhacked roms.

cooldude413, 2017-06-21:
Clean unhacked roms crash sometimes, much less often than hacked ones; but still crash

DeathAngel74, 2017-06-21:
I would try another source for roms.

cooldude413, 2017-06-21:
I've tried several different sources, same ones that fail on my 3ds sometimes, work perfectly on my gfs 3ds (she has same Luma version on o3ds)

Hikari-chin, 2017-06-22:
I've got a New 3DS as well, and the reproduce this. Try a new SD card or formatting your existing one to use 32K clusters (64K is known to cause AGB issues).

closed 2017-06-22 by Hikari-chin

cooldude413, 2017-06-24:
It was because of the type of the SD card (SP, 128gb,32k cluster). I switched to Samsung and now it works every time. Thanks Hikari-chin and Death Angel!!!
Really appreciate the help :)