LumaTeam/Luma3DS #643: Region Emulation doesn't work with Puyo Puyo Tetris and Puyo Puyo Quest DLC properly

internetakias, 2017-06-21:
**System model:**

New 3DS

**SysNAND version (+emu/redNAND version if applicable):**

11.4.0-37E SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap

**Luma3DS version:**

8.0 Stable

**Luma3DS configuration/options:**

Default EmuNAND: (1)

Screen brightness: (1)

Splash: (Off)

PIN lock: (Off)

New 3DS CPU: (Off)

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Use custom path: ( )

Enable game patching: (X)

Show NAND or user string in System Settings: (X)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

--

**Explanation of the issue:**
Region emulation does not work properly with Puyo Puyo Tetris's and Puyo Puyo Quest's DLC. In Puyo Puyo Tetris, you can access the extra chapters - but you can do that without region emulation enabled also - but not the added BGM (selecting them will just play back the default battle theme), alternate character voices or piece skins. Chronicle is the same as far as the BGM goes (except selecting the added BGM doesn't play back anything). Not sure about everything else.

**Steps to reproduce:**
Note: This is for Puyo Puyo Tetris

1.Enable Game Patching in Luma settings

2.Go to SD:/luma/titles/0004000000101200

3.Create locale.txt with the following text inside: "JPN JP JP 01"

4.Repeat steps 2 and 3 for the update (Title ID: 0004000E00101200. Not sure if the process has to be repeated for the DLC too. Though, I've done it and nothing's changed.)

5.Launch the game.

6.Press A on title screen.

7.Press A.

8.Press A.

9.Press A.

10.Select characters.

11. Select the first button from the bottom.

12.Go down to the "BGM" option

13. Select any song starting from 45.

**Dump file:**

Hikari-chin, 2017-06-22:
Could you try a later commit like [this one](https://astronautlevel2.github.io/Luma3DS/builds/Luma3DS-1782827.zip) and let us know if the issue still occurs?

internetakias, 2017-06-22:
Nope. By the way, the description for game patching says that the archive patch is required. Does that mean I have to enable that through the rosalina menu?

Jan200101, 2017-06-27:
I myself tried Puyo Puyo Tetris on a eu system and it worked fine

internetakias, 2017-06-27:
Does that include the DLC too? Can you use all the DLC music, voices and skins properly?

TuxSH, 2017-07-02:
I guess it works on the 8.1 release and/or it's user error if it isn't?

Hikari-chin, 2017-07-02:
Gonna close this. Has gone quite awhile without a response/further confirmation. Will reopen if necessary.

closed 2017-07-02 by Hikari-chin

internetakias, 2017-07-04:
It still doesn't work, even on 8.1 stable. Also, I don't see how it could be user error. I've listed what the locale.txt files contain up above and as far as I'm aware it's all correct and similar to what I've put in locale.txts for games that actually work. By the way, I noticed that the DLC BGM issue seems to behave differently depending on the mode. With VS and Party, selecting a DLC BGM track plays the default music instead. With Fusion, however, no music plays at all.

astronautlevel2, 2017-07-06:
Confirmed to not be a Luma problem, as it doesn't work on a Japanese NAND either.

ramdompersom, 2021-02-11:
This also happens to me.
Seens like if you use the translated dlc it works fine.