LumaTeam/Luma3DS #644: Games Crash (mostly Fire Emblem Awakening)

arcenedonoob, 2017-06-21:
Issue: Everytime I go into battle in Fire Emblem Awakening, it gives me the error:
An exception occurred
Processor: ARM11 (core 0)
Exception type: data abort
Current process: quartz (00040000000A0500)

R0 00000000
R2 00000005
R4 00000000
R6 00000004
R8 004D1120
R10 000000FF
R12 00000001
LR 002C22F0
CPSR 60000010
R1 00000000
R3 15915968
R5 00000100
R7 00000000
R9 00000000
R11 004D1100
SP 0FFFFEC8
PC 002EBE80
FPEXC 00000005

Please help!

I have a:

o3DS XL
11.4.0-37U SysNAND
Boot9Strap
v8.0 stable

Default EmuNAND: (y)

Screen brightness: (4)

Splash: (before)

PIN lock: (no)

Autoboot EmuNAND: (yes )

Use EmuNAND FIRM if booting with R: (no)

Enable game patching: (yes)

Show NAND or user string in System Settings: (no)

Show GBA boot screen in patched AGB_FIRM: (yes)

Patch ARM9 access: (no)

Set developer UNITINFO: (no)

HenryLeonheart, 2017-06-22:
I used to have that error when I was using NTR's cheat plugin that made the characters have infinite movement. When they pair up, the characters even though paired up, are separate on the map. 

When you go to a map without both of them, you get an error like that. If this is your problem, then go to a free battle (against Risen for example) and separate them.

arcenedonoob, 2017-06-22:
Everytime I go against the Risen it still crashes.

closed 2017-06-22 by arcenedonoob

ammomammal, 2017-06-22:
i have the same issue

snaptastica, 2017-06-23:
I also have this issue. I played a LOT of fe:a on previous versions of Luma and never had it before

glitchika32, 2017-11-03:
the error appears even when you dont have NTR just pop up at random,making the game UNPLAYABLE

Link5261, 2018-10-27:
I got this error upon _closing_ the app after having had it open for a few hours using the Rosalina menu to activate some cheats. The cheats worked just fine and included 'trade with enemies', 'every action worth 100xp', 'infinite movement (turn never ends)', 'control enemies', and 'every stat grows upon level up'. It didn't cause any data loss or any issue, seeing as I was closing the game anyway, but just wanted to look it up and found this issue posting.

n2DS XL
11.8 sysNAND
BootStrap9
Luma 9.1-bd15f51

All I can remember from the error screen was:
Processor: ARM11
Current process: quartz