LumaTeam/Luma3DS #647: Multiple games crashing (I can temporarily stop it by reinstalling the cias but it seems like it happens after 3 times of playing a certain game,then it just becomes a timed thing after booting a game)

ExcaliburAmi, 2017-06-21:
**System model:** O3DS

SysNAND version (+emu/redNAND version if applicable): 11.4.0-37E Sys , no emunand

Entrypoint (How/what you're using to boot Luma3DS): Boot9Strap

Luma3DS version: 8.0 Stable

Luma3DS configuration/options:

Default EmuNAND: ( 1)

Screen brightness: (4 )

Splash: (Off )

PIN lock: (Off)

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Use custom path: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (X )

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

--

Explanation of the issue: 
After updating to boot9strap and luma 8.0 I've been experiencing a lot of "an error has occured"s,and this is different from the ARM11 crashes,it's a crash that's caused by some process (in the process list) and every game has its own process that crashes,but how this happened is that after importing some old saves and reinstalling the cias the games work fine,but afterwards they just crash,also this only happens with games that I imported saves into (also this is after I unbanned myself and made a new Mii).
here's what it looks like (this is for Kid Icarus Uprising) : 
http://gbatemp.net/attachments/20170620_024726-jpg.90826/

Steps to reproduce:
1.Switch your friendcode (friendcodeseed b)
2.Import an old save from before you switched the friend code
3.Play the game for a while (3 times or more)
And it crashes
I don't think this is reproduceable but for anybody who wants to give it a shot go for it

ExcaliburAmi, 2017-06-22:
actually that's from super mario 3D land (the one from kid icarus uprising shows "apollo" instead of "CtrApp")

OmeletHopper, 2017-06-22:
I used to have a similar problem, and I nailed it down to my SD card. I had been using a 16GB one after I installed A9LH + Luma, and I started noticing problems in installed games on my SD. You said you can temporarily fix it by reinstalling the .cia, which leads me to believe your SD is corrupting your data.

ExcaliburAmi, 2017-06-22:
Yeah but having to reinstall the cia everytime just to have it die happen again  is too much

Hikari-chin, 2017-06-22:
Can't reproduce this. Very likely a bad SD card. Format it and if that doesn't fix it, buy a new one.

closed 2017-06-22 by Hikari-chin

Margen67, 2017-06-22:
Try h2testw. Backup your files first.

ExcaliburAmi, 2017-06-22:
well,formatted my sd card,forgot I don't have CTRNAND Luma,now i don't know what to do because I can't even power on my 3ds anymore

Margen67, 2017-06-22:
Just put the boot.firm for Luma on the SD card.

ExcaliburAmi, 2017-06-24:
Actually the crashes happen more if you format the sd card using godmode9,so yeah that's weird,so after formatting my sd card through pc the crashes started happening way less now and at random times

ExcaliburAmi, 2017-06-24:
By the way OmletHopper what did you do to fix the problem ?

Margen67, 2017-06-24:
Probably by buying a new SD card.

@ExcaliburAmi Test your SD card with h2testw.

ExcaliburAmi, 2017-06-24:
@Margen67 When I try to backup my files a file named 000000.app gives me an error like "couldnt read file from disk" and it inconviently makes me not able to back them all up without the fear of screwing something over, so now I cant fully test my sd card

Hikari-chin, 2017-06-25:
So it should be clear now that this isn't a Luma issue, it's a hardware one, and this isn't a forum for tech support. Locking.