LumaTeam/Luma3DS #648: Entering or exiting sleep mode while Rosalina menu is up doesn't work

vaguerant, 2017-06-22:
**System model:**

New 3DS (XL)

**SysNAND version (+emu/redNAND version if applicable):**

11.1.0-34U SysNAND, 9.2-something EmuNAND (no longer actively in use)

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap 1.2

**Luma3DS version:**

8.0 stable

**Luma3DS configuration/options:**

Default EmuNAND: ( 1 )

Screen brightness: ( 2 )

Splash: ( Off )

PIN lock: ( Off )

New 3DS CPU: ( Clock+L2 )

--

Autoboot EmuNAND: ( Off )

Use EmuNAND FIRM if booting with R: ( Off )

Enable loading external FIRMs and modules: ( Off )

Use custom path: ( ? does this feature even exist any more, I don't see it or use it )

Enable game patching: ( On )

Show NAND or user string in System Settings: ( On )

Show GBA boot screen in patched AGB_FIRM: ( On )

Patch ARM9 access: ( Off )

Set developer UNITINFO: ( Off )

--

**Explanation of the issue:**
The Rosalina menu suppresses both entering and exiting sleep mode.

**Steps to reproduce:**
**Entering sleep mode:**
1. Open Rosalina menu using defined shortcut

2. Close 3DS (presumably also works with 2DS sleep switch but I don't know one of those) and notice 3DS is not sleeping (light from screen is still visible in low light environments), etc.

**Exiting Sleep Mode:**
1. Sleep the 3DS (have Rosalina menu NOT open)
2. Enter the Rosalina shortcut (probably easier to do if you change the shortcut to e.g. L+R first, but I'm able to hit L+D-Pad Down + Select with the N3DSXL "almost closed" and keep it sleeping,
3. Re-open (un-switch on 2DS?) the 3DS and notice 3DS is still sleeping.
4. Press B to exit Rosalina, which wakes the 3DS.

This is a dumb bug, I don't think people are likely to encounter it often.

fox8091, 2017-06-22:
Confirmed on 2DS. Actually, the system remains fully responsive in the Rosalina menu, and crashes to sleep mode upon exit. Worse yet, you can't "unsleep" it by pressing B to exit Rosalina.

vaguerant, 2017-06-23:
Coming at this from another direction: Since discussing it with TuxSH, I suppose the real bug here is not the suppression of sleep entry/exit (a result of intended behavior, Rosalina blocking other threads), but the fact that the Rosalina menu can be entered during sleep mode. Unless there's a reason this is useful, perhaps the real solution would be to add a condition that the 3DS is not sleeping in addition to the Rosalina shortcut, such that Sleep + L + Down + Select (or user's preferred shortcut) == no action; Awake + L + Down + Select == trigger Rosalina.

fox8091, 2017-06-23:
Another idea to complement yours: Perhaps close Rosalina if it detects the console trying to enter sleep mode?

Spodi, 2017-06-24:
Can confirm that opening Rosalina prevents sleep. But I can't open "her" while already in sleep mode on my N3DS (using a magnet). Using v8.0-17828273

WubTheGame, 2017-06-26:
@fox8091 that is to go in the Rosalina suggestions page. found here: (will place a link here in a moment)

TuxSH, 2017-06-30:
> Probably Rosalina freezes all processes before sleep mode is properly disabled.

Apart from sysmodules except gsp, it does.

And I think it's actually working in sleep mode, just not displaying anything.

DMSalesman, 2017-08-03:
Bug confirmed on Old 3DS. If Rosalina is entered while the device sleeps, pressing B to wake it on screen opened does not work.

SirNapkin1334, 2018-02-01:
Yes. This happens to me. I figured it was intentional.

SirNapkin1334, 2018-02-01:
(can confirm on O3DSXL, O3DS, and N2DSXL)

TuxSH, 2020-04-15:
Closing, tracker cleanup. Known issue, tried to fix it multiple times to no avail

closed 2020-04-15 by TuxSH