LumaTeam/Luma3DS #650: boot9strap Rosalina menu crashes the system when plugged in

NLDoA, 2017-06-22:
**System model:** 

2DS

**SysNAND version:**

11.4.0-37U

**Entrypoint:**

boot9strap 1.2

**Luma3DS version:**

8.0

**Luma3DS configuration/options:**

Default EmuNAND: (1)

Screen brightness: (4)

Splash: (Off)

PIN lock: (Off)

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Use custom path: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (X)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

--

**Explanation of the issue:**

So, I was playing Sun on my 2DS (which does have to run in extended memory mode, Idk if that makes a difference) and I opened the Rosalina menu. I noticed my battery was getting a little low, so I plugged in the charger. It made both screens immediatly go into a striped color, and crashed the game. The only way to restore it was to hard power off the DS. No data was corrupted (as far as I can tell), and unfortanatly, now I can't reproduce it. I didn't have the dump file setting on either, so it didn't create one. The only modification for the game I have is no borders, but I don't think that is the issue, especially since I can't reproduce it anymore.

Hikari-chin, 2017-06-22:
The screen corruption is a known issue and is being worked on. Has nothing to do with being plugged in or extmem mode, though, it's just random.

closed 2017-06-22 by Hikari-chin

Hikari-chin, 2017-06-25:
This should be fixed in commit 2363817.