LumaTeam/Luma3DS #651: Animal Crossing New Leaf JPN not possible to play

Seb9997, 2017-06-22:
**System model:**

New 3DS XL

**SysNAND version (+emu/redNAND version if applicable):**

11.4.0-37E SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap/Sighax

**Luma3DS version:**

Luma3DS v8.0

**Luma3DS configuration/options:**

Default EmuNAND: ( 1 )

Screen brightness: ( 4 )

Splash: ( Off )

PIN lock: ( Off )

New 3DS CPU: ( Off )

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: ( X )

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

**Explanation of the issue:**

Everytime if i try to start the japanese version of Animal Crossing New Leaf i get the message "An exception occured".
If i try it with the japanese version of Pokemon X everything works fine.
Both are cards i ordered from Japan

**Steps to reproduce:**

1. try to start the game

2. wait til the message appears

**Dump file:**
[dump.zip](https://github.com/AuroraWright/Luma3DS/files/1096374/dump.zip)

ZenIsBestWolf, 2017-06-23:
Attempt using Homebrew. Use smea region unlocker.

HugaShire, 2017-06-23:
Turn on Game Patching and then use LumaLocales to make a txt file to change the region to JPN

closed 2017-06-25 by Hikari-chin

Seb9997, 2017-07-01:
Thanks a lot Huga Shire it really worked