LumaTeam/Luma3DS #652: Ever Oasis EUR causes error every time you turn on 3DS if you go to manual screen

panchito714, 2017-06-23:
**System model:**

New 3DS XL

**SysNAND version (+emu/redNAND version if applicable):**

11.4.0-37U SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap/Sighax

**Luma3DS version:**

Luma v8.0.174089e4

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: (4)

Splash: (off)

PIN lock: (off)

New 3DS CPU: (off)

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Use custom path: ( )

Enable game patching: (X)

Show NAND or user string in System Settings: (X)

Show GBA boot screen in patched AGB_FIRM: (X)

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

--

**Explanation of the issue:**
If you press start to open in game menu in Ever Oasis and press x to open the manual it will give you an error and tells you to press any button to shutdown. Then every time after that if you try turning on your 3ds it will keep giving you that error and not allow you to go to home menu screen of your 3ds.

**Steps to reproduce:**

1. Play game until it lets you go into the start menu.

2. Press X to go to manual.

**Dump file:**
[crash_dump_00000036.zip](https://github.com/AuroraWright/Luma3DS/files/1096713/crash_dump_00000036.zip)

ZenIsBestWolf, 2017-06-23:
You say your firmware is USA, just try the USA version first.

angelmarlboro, 2017-06-24:
can you upload a picture of that error?

Kalanyr, 2017-06-25:
I just tried the US EShop version of Ever Oasis on a US N3DS and the manual works fine.

panchito714, 2017-06-25:
This was EUR version on a USA N3DS not sure if only happens with EUR version.
Got it working again, deleted everything, reinstalled all the games and copied over their save files.
Picture of error screen https://m.imgur.com/a/etbMx

Hikari-chin, 2017-06-25:
This is possibly s duplicate of #150.

Link2738, 2017-06-26:
The EUR game has been causing problems on out of region systems including a gamebreaking glitch several hours in, Use a US .3ds/.CIA for the time being

Hikari-chin, 2017-06-26:
So, this is **not** a duplicate of #150. I just downloaded the EUR version of Ever Oasis and configured it to use locale emulation (otherwise it'd crash on launch), and can launch the manual just fine without any crashes.

closed 2017-06-26 by Hikari-chin

makkusu19, 2017-07-01:
So my N3DS was working fine. So when I tried to play this game it said, ERROR. So I tried it again, and then my 3ds just randomly shut off. It won't even turn on anymore. I thought it was the popping shit, because now after I turn it on, the blue light stays on for a few seconds, then it makes a pop, and then shuts off. So I took it to a repair place and they couldn't fix it, but I noticed when I take out the sd card it turns on for like a second then shuts off so it must be a software problem right?

Hikari-chin, 2017-07-01:
@makkusu19 Ever Oasis only requires locale emulation to work correctly; there's no feasible way that it or Luma could've caused that issue. Additionally, that your device shuts down without its SD card means that your it is still managing to get far enough for boot9strap. 

What else could've caused this, I have no idea. Sounds like an unfortunate coincidence. However, the Luma issues aren't for tech support, so please try the [NH Discord](https://discord.gg/MWxPgEp).