LumaTeam/Luma3DS #653: "An error occurred (ErrDisp)" with wifi turned on

jorgarsan, 2017-06-23:
**System model:**

Old 3DS

**SysNAND version (+emu/redNAND version if applicable):**

11.3.0-36E SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap 1.2

**Luma3DS version:**

Luma 8.0

**Luma3DS configuration/options:**

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

PIN lock: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Use custom path: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

--

**Explanation of the issue:**
After install Luma 8.0, FBI and HBL are working ok, but when the wifi is turned on, console freezes in 15-20 seconds. If the wifi is turned off, the console is working fine.

**Steps to reproduce:**

1. Start the console

2. Turn on wifi

**Dump file:**
An error occurred (ErrDisp)

Error type: generic
Process ID: 25
Process name: num
Process title ID: 0x0004012000002d02

Adress: 0x0010941a
Error code: 0xf9606c02

Press any button to reboot

closed 2017-06-23 by jorgarsan

gicosta2016, 2017-11-16:
solution: replace wifi board!!!!!

betelgeuse777, 2018-03-04:
dont need to replace wifi board
just open the nintendo quit the wifi board and connect it again 
is just a contact problem between the boards
i had the same problem