LumaTeam/Luma3DS #654: i cant enter Luma 8 menu

angelmarlboro, 2017-06-24:
hello people, if i holding start button and press power, and still holding start button the new 3ds xl boot normal, not luma´s menu.

what happen here? all games run fine

thanks

Trojaner1982, 2017-06-24:
Luma Menue = Hold Select + Press Power
Start Button is for progs like Godmode9, Decrypt9

if Godmode, Decrypt doesn´t start check te Luma/payload folder if the .firm files are there (ex: Luma/payload/godmode9.firm ...

angelmarlboro, 2017-06-24:
thanks dude. thast right my mistake.
i have luma 8.0 and my problem is very random, 3 times from 10 the nintendo dosn´t power off, i need press power button 15 second to power off. can you help me anyone with that?

Thanks again Trojaner1982.

How do I know the rest of the installed versions and update them?
is there a new version of luma?
is there a new version of godmode?

Trojaner1982, 2017-06-24:
8.0 is still the last until now
if you like you can get here always the Nightly Builds [https://astronautlevel2.github.io/Luma3DS/](url)

Godmode9 = [https://github.com/d0k3/GodMode9/releases](url)
Decrypt9 = [https://github.com/d0k3/Decrypt9WIP/releases](url)
you need the .firm files from this in the Luma/payloads Folder, not the bin anymore

angelmarlboro, 2017-06-24:
Thanks Trojaner1982 but when i click in the url says " this is not web page you are locking for " is the link ok?

Trojaner1982, 2017-06-24:
no sorry, have that power issue too sometimes ... but it´s not a big problem,

angelmarlboro, 2017-06-24:
You never had the problem arm11 crash with any app? shop app? games?

Trojaner1982, 2017-06-24:
only one time ... Every APP/Game crashed with ARM Message, or a nice other error "No SD Card" ... restored my Clean Nand Backup, fresh formated the SD, installed all apps/games new and no proplem with crashes anymore

angelmarlboro, 2017-06-24:
where i can find a tutorial to backup my NAND or where shoud it be? to backup your games and savegames copy paste in folder in windows right?

Shadow322, 2017-06-24:
This is not a help place ask your question on gbatemp

closed 2017-06-24 by TuxSH