LumaTeam/Luma3DS #655: Launching FBI from HBL 2.0 triggers an arm11 exception

chenzw95, 2017-06-24:
**Luma3DS configuration/options:** (per 3ds.guide recommendation)

Default EmuNAND: ( )

Screen brightness: ( )

Splash: ( )

PIN lock: ( )

New 3DS CPU: ( )

--

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Use custom path: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: (X)

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

--

**Explanation of the issue:**

There has been a steady stream of users in the Nintendo Homebrew channel who encounter arm11 exceptions when attempting to launch FBI from the HBL.

**Steps to reproduce:**

1. Launch FBI from HBL 2.0 (seamless loading).

**Dump file:** Multiple attached, collected from various users on the Nintendo Homebrew server: 
[crashdumps.zip](https://github.com/AuroraWright/Luma3DS/files/1099440/crashdumps.zip)

Trojaner1982, 2017-06-24:
No Crashes here, try the last Nightly [https://astronautlevel2.github.io/Luma3DS/builds/Luma3DS-0377cbd.zip](url)

Hikari-chin, 2017-06-24:
O

closed 2017-06-24 by Hikari-chin