LumaTeam/Luma3DS #657: Luma V8 Doesnt support target titles

ShadowMew718, 2017-06-25:
I recently got Luma only to find out I cant run stuff so can someone tell me a fast and easy way to get titles, I don't want no copy and pasted crap from other closed issues because I have no actual clue what they mean and all I want is to run my dang apps I'm on 11.4 2ds

Margen67, 2017-06-25:
If you want help you can start by being more specific, and using the issue template 😛

ShadowMew718, 2017-06-25:
Sorry I read most of it but I couldn't answer it because i am doing something on GodMode9 so I cant really check back and answer everything and as I said I don't really know anything about Luma and stuff I just want my titles to work so I can do stuff but is there anyway I could get them to work, if I loaded HB without Luma it would work right? That's how I got Luma in the first place so Is there anyway I can get another exploit for 11.4 or am I totally stuck till something like another firmware update comes or if stuff like Soundhax works again because it got patched in 11.4. (I am really sorry for just making an issue and haven't provided any real details)

ShadowMew718, 2017-06-25:
System model: 2DS

SysNAND version:11.4.0-37U

Entrypoint: Think its Boot9Strap

Luma3DS Version:8.0

Luma3DS configuration/options:

Default EmuNAND: 1(X)
Screen brightness: 4(X)
Splash: Off(X)
PIN lock: Off(X)

Autoboot EmuNAND: ( )
Use EmuNAND FIRM if booting with R: ( )
Enable loading external FIRMs and modules: ( )
Use custom path: ( )
Enable game patching: ( )
Show NAND or user string in System Settings: (X)
Show GBA boot screen in patched AGB_FIRM: ( )
Patch ARM9 access: ( )
Set developer UNITINFO: ( )

Explanation of Issue: Luma wont let me run target titles because its not supported.

Hope that's enough. :)

fincs, 2017-06-25:
The answer is copied and pasted for a reason: it contains all the relevant substitutes for any usages of target titles that you may have. Target titles are *not* going to be implemented.

>This is intentional. It was decided that target title selection is not useful anymore since **3dsx applications have full ARM11 system permissions now**, and it is possible to directly access the savefiles of any title by using the appropriate FSUSER commands. In the meantime, you can use the CIA versions of savegame managers.
>
>Also, HANS is obsoleted by Luma3DS's built-in ExeFS/RomFS (LayeredFS) game patching feature. You can also use GodMode9 instead of braindump for dumping game assets.

In short:

- Savegame management: Use JKSM
- Romhacks (HANS): Use Luma's built-in feature
- Game dumping: Use GodMode9

closed 2017-06-25 by fincs