LumaTeam/Luma3DS #658: Cant turn 3ds on.

hii915, 2017-06-25:
Hi, I successfully updated a9lh to b9s.  But in the process of updating b9s and luma my 3ds's blue led turns on for a second then turns off.  This happened to me in section 5 of Updating B9S.  I tried updating luma to 7.1 and 8.0.  I tried booting with no SD and same issue.  Also I believe this happened section V of updating B9S steps 13-22.  Help will be appreciated thanks!  

_Sent from my Huawei Nexus 6P using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

ValiantBlade, 2017-06-25:
Try using an SD with Luma 7.1 on it. Did you install B9S 1.0 or 1.1 or 1.2?

DeathAngel74, 2017-06-25:
Use Luma 7.0.5/7.1 for B9S 1.0/1.2. Luma 8 requires B9S 1.2 for the 3DS to get to the home screen.

hii915, 2017-06-25:
I installed b9s 1.0 and then updated luma to 8.0.  Both 7.1 and 8.0 are unbootable

_Sent from my Huawei Nexus 6P using [FastHub](https://play.google.com/store/apps/details?id=com.fastaccess.github)_

HugaShire, 2017-06-25:
https://3ds.guide/troubleshooting#ts_sys_b9s

Hikari-chin, 2017-06-25:
GitHub issues aren't for tech support, sorry guys.

closed 2017-06-25 by Hikari-chin