LumaTeam/Luma3DS #659: Pokemon Red/Yellow/(Blue?) cause Prefetch Abort crash

N-G-Neer, 2017-06-25:
System model : Old 3DS

SysNAND version : 11.4.0-37E SysNAND

Entrypoint : Boot9Strap

Luma3DS version : 8.0 stable

Luma3DS configuration/options: Default EmuNAND 1, Screen Brightness 4, Splash Before, PIN off, Show NAND or user string.

Explanation of the issue : 

Downloaded and installed Pokemon Red and Yellow (GBA) on Freeshop.
Both initially crashed at the Nintendo 3DS logo with the following error : 
An exception occurred.
Processor: ARM11 (core1)
Exception type: prefetch abort (svcBreak)
Current process: loader
(dump file included)

After deleting and reinstalling them a few times, I got Yellow to work while Red still crashes.

Steps to reproduce : 
1. Download Pokemon Red or Yellow on Freeshop (French Versions at least)

2. Launch the titles once installed.

[crash_dump_00000000.zip](https://github.com/AuroraWright/Luma3DS/files/1100719/crash_dump_00000000.zip)

j0zu3, 2017-06-26:
This happen to me with my own dump of mhgen, same exception tiype and same current process error, I had to do a new dump and reinstall a few times until it worked

Hikari-chin, 2017-06-26:
Can't reproduce this, but also: We don't and won't provide support for issues pertaining to piracy tools here.

closed 2017-06-26 by Hikari-chin