LumaTeam/Luma3DS #660: Mario party island tour crash arm 11

killer983, 2017-06-26:
**System model:**

New 3ds xl

**SysNAND version (+emu/redNAND version if applicable):**

11.4.0-37E SysNAND

**Entrypoint (How/what you're using to boot Luma3DS):**

Boot9Strap/Sighax 1.2

**Luma3DS version:**

8.0 stable

**Luma3DS configuration/options:**

Default EmuNAND: (1 )

Screen brightness: (4 )

Splash: (off)

PIN lock: ( off)

New 3DS CPU: (off)

Autoboot EmuNAND: ( )

Use EmuNAND FIRM if booting with R: ( )

Enable loading external FIRMs and modules: ( )

Use custom path: ( )

Enable game patching: ( )

Show NAND or user string in System Settings: ( )

Show GBA boot screen in patched AGB_FIRM: ( )

Patch ARM9 access: ( )

Set developer UNITINFO: ( )

**Explanation of the issue:**
While playing the Mario party island tour I jumped a screen with letters and numbers while playing a mini game

**Steps to reproduce:**

1.
to enter the game

2.
Play a mini game

**Dump file:**
[crash_dump_00000000.zip](https://github.com/AuroraWright/Luma3DS/files/1101867/crash_dump_00000000.zip)

HugaShire, 2017-06-26:
Try reinstalling the game. It could be the game files being corrupted.

killer983, 2017-06-26:
Was installed with freeshop not by cia. You can also corrupt files

ZenIsBestWolf, 2017-06-26:
Doesn't happen to me on USA.

closed 2017-06-26 by TuxSH