LumaTeam/Luma3DS #661: IPS patch is not being applied

Insighter2k, 2017-06-26:
New 3DS
11.4.0-37E SysNAND
Boot9Strap 1.2
Luma Version 8.0 (Stable)
Enable game patching: (x )
Show NAND or user string in System Settings: ( x)

I am working on a current translation. The base game has been patched and a new code.bin file has been added as well. the new code.bin file needs to be edited via an ips-patch file.

If  i am adding the modified code.bin file, it does work.
it is not working if i am putting the ips file under the title folder.

Could it be that the ips file patches only the base game code.bin file but not the new one from the patch?

fox8091, 2017-06-27:
Wait, so you have both a code.bin and a code.ips in your Luma folder?

Insighter2k, 2017-06-27:
No. I am putting either the code.bin or the code.ips file in.

DeathAngel74, 2017-06-27:
You should have the code.bin or code.ips in the luma/titles/*TITLEID*/ folder. Forgive my ignorance, if you've already tried this.

Insighter2k, 2017-06-27:
I did. As I have written, using the modified code.bin works, but not the ips.

vaguerant, 2017-06-27:
Long shot: Is there any chance that you have file extensions hidden on a platform (like Windows, by default) that does that and your patch is actually named "code.ips.ips"?

ericjwg, 2017-06-27:
does onionfs work with that ips?

Insighter2k, 2017-06-27:
No file extension hidden. I enabled extension by default.

I havent tested onionfs yet.

Hikari-chin, 2017-06-28:
Could you try putting **both** the code.bin and code.ips in the folder at the same time?

architdate, 2017-06-29:
What game/patch is this? since for sun and moon the ips patches seem working perfectly

piratesephiroth, 2017-06-29:
>the base game has been patched and a new code.bin file has been added as well
You mean an update title was installed?
If that's the case then I think you should be adding an ips to that update's code.bin (title id low 0004000e) because it is the one launched instead of the original one from the game.

ericjwg, 2017-06-29:
it's the code.bin from update patch @piratesephiroth

piratesephiroth, 2017-06-29:
Well then it only patches the main code.bin.
Perhaps you can try loading the patched update code.bin to replace the main code.bin?

Insighter2k, 2017-06-29:
Sorry for the delay.i will be able to test onion on the weekend.

The modified code.bin from the patch/update works flawlessly. And I am using this method currently. Haven't tried the original code.bin plus ips patch, but then I could use the modified file from the beginning.

Hikari-chin, 2017-07-05:
Were you able to test OnionFS? 

If you still haven't got this working, could you please provide the relevant code.ips file, please?

Hikari-chin, 2017-07-07:
Going to close this for now. Once the code.ips and .bin are provided and if it becomes apparent this is an issue with Luma, I'll reopen it.

closed 2017-07-07 by Hikari-chin

ericjwg, 2017-07-07:
https://puu.sh/wDN4h/87f2483c1b.ips

Hikari-chin, 2017-07-08:
@Insighter2k I believe I've found your issue here. This game has an update/patch available for it, 1.2. While code.bin would overwrite the update code.bin entirely, code.ips is specific to a single version of the code.bin, so if your code.ips is based on a 1.0 code.bin it wouldn't work with 1.2 installed (or vice versa). 

[Here](https://u.nya.is/wmmavq.ips) is a code.ips based on the latest patch/updated code.bin for that game. Could you test it?

Insighter2k, 2017-07-09:
I made several changes to the 1.2 code.bin. I will create a new ips patch, test it and post the result. Also might post the ips itself. The posted ips could break the game, because of the ongoing changes.

Btw, in case of performance, does the ips feature takes a lot of performance or isn't it noticeable?

AuroraWright, 2017-07-09:
Take note that you need to use the patch's code.bin as a base for the ips patch (you need to dump it from the update) and that it will only work if users have the patch installed

AuroraWright, 2017-07-09:
That said I tested the code.bin on the gbatemp thread and the code.ips I created and I noticed no differences, although I didn't know where to look

Insighter2k, 2017-07-09:
it did work. i checked with the tenkaichi part of the game which had some changes with the 1.2 patch.

what about the performance? i havent noticed not a huge difference, but is there?

AuroraWright, 2017-07-09:
The code is patched before the game even starts, so there are no difference (patching is very quick). ips patches were mostly implemented to allow people not to have to provide full code.bins (as that's technically illegal).