LumaTeam/Luma3DS #663: Hatsune Miku Project Mirai DX US romfs patching not working

Link2738, 2017-06-26:
**O3DS**

**11.3.0-36U**

**Boot9Strap):**

**8.0 Stable**

**Luma3DS configuration/options:**

Splash: ( x)

Enable game patching: (x )

Show NAND or user string in System Settings: (x )
--

**Explanation of the issue:**
A made a custom texture for the game,but it lags horribly to start and doesn't patch the game

(I've striped the romfs to the single file and it still lags)

**Steps to reproduce:**

1.Get a .3ds/.CIA of the game

2.Get the romfs folder

3.Change clothing or map textures (You should know how this works)

4.Reinject and create folder directory on the SD

5.Start game with patching enabled

Hikari-chin, 2017-06-26:
Can't reproduce this; make sure you're [using the correct directory structure](https://github.com/AuroraWright/Luma3DS/wiki/Options-and-usage), that your modifications are valid for the version of the game you have installed, etc. If that's the case, please try a newer nightly [like this one](https://astronautlevel2.github.io/Luma3DS/builds/Luma3DS-d42e938.zip) to see if it works. If not, please try an alternative LayeredFS patching implementation like OnionFS and report back if it works.

Link2738, 2017-06-26:
I changed textures for other games so I know the directory structure but i'll try your solutions

Link2738, 2017-06-26:
Nothing worked sadly did it work for you?

Hikari-chin, 2017-06-26:
Yes. That points to user error, or rather: this is very probably not a Luma issue.

closed 2017-06-26 by Hikari-chin

Link2738, 2017-06-26:
Long story short it was the .CIA I used was slightly bad, it was still playable but the romfs  extractor had some problems I got another one and it works now. Thanks for trying to help and GJ with luma.